thingsboard/thingsboard-gateway #1413: [HELP]

Godsgrav3, 2024-05-29:
**Describe the issue**
Hi, I'm currently testing thingsboard out with its version control system (linked to our gitlab environment), and I have been looking for a way to backup/version control the MQTT connector configuration in the gateway, but I can't seem to find any way to do this. Is there a way to version control the connector configuration via the web interface?

I have found the connector configuration but its in a secure docker volume (can access it with sudo, but generally don't want to use sudo) at "/var/lib/docker/volumes/tb-gw-config/_data". Also according to the documentation "https://thingsboard.io/docs/iot-gateway/configuration/" the config data should also be here "/etc/thingsboard-gateway/config", but the folder doesn't exist. could be explained because I use the docker compose gateway yaml (downloaded from the web interface), but would like to know if this is an issue or not.

**Versions (please complete the following information):**
 - OS: Ubuntu 23.10
 - Thingsboard IoT Gateway version 3.5
 - Python version 3.11

imbeacon, 2024-05-29:
Hi @Godsgrav3,

Thank you for your interest in ThingsBoard IoT Gateway.
At the moment you can get configuration only in docker volume or you may find them in attributes of the gateway device. Unfortunately, at the moment there is no way to save configuration using version control feature, but we will discuss this point and maybe it will be available in the future. Unfortunately, I cannot guarantee this, because of version control feature is designed to save entities states, not their attributes and telemetry.

Godsgrav3, 2024-05-29:
HI @imbeacon,

Thank you for the information! I am looking forward to this being implemented.

closed 2024-05-29 by Godsgrav3