thingsboard/thingsboard-gateway #1416: Gateway Related Issus- “tb-gateway | exec /bin/bash: exec format error”

SecurityEngineers, 2024-05-31:
I am trying to make a Thingsboard IoT gateway on my Raspberry Pi 4 with latest Raspberry Pi OS (Bookworm) but I am getting an error- `tb-gateway | exec /bin/bash: exec format error`

I setup Docker and Docker Compose correctly on my RPi, created a new gateway in my Thingsboard account and downloaded the .yml file. Then when I try to run this code using `sudo docker compose up`, get error (It's working in my windows-wsl but not in RPi).

(I also followed this links below-
https://thingsboard.io/docs/iot-gateway/getting-started/
https://thingsboard.io/docs/iot-gateway/install/docker-installation/)

So please someone help me to do this.
Thank You

[image]

imbeacon, 2024-06-06:
Hi @SecurityEngineers,

Thank you for your interest in ThingsBoard IoT Gateway.
Could you show output of command on your RPI: ```docker inspect thingsboard/tb-gateway```

imbeacon, 2024-06-20:
Hi @SecurityEngineers,

We have created a separated image for arm/v7 platform, please try to use tb-gateway:3.5.1-rpi instead of the latest image.

closed 2024-06-20 by imbeacon