thingsboard/thingsboard-gateway #1422: [BUG]

ZhaoLinl, 2024-06-11:
**Describe the bug**
This problem is displayed after the Socket connection is established and the device sends data to the TB-gateway.
[image]

**Connector name (If bug in the some connector):**

**Error traceback (If available):**
```
'deviceName'
Traceback (most recent call last):
 
**Versions (please complete the following information):**
 - OS: [e.g. Ubuntu 18.04]
 - Thingsboard IoT Gateway version [3.4.6]
 - Python version[e.g. 3.9]

samson0v, 2024-06-11:
Hi @ZhaoLinl, thanks for your interest in ThingsBoard IoT Gateway!
Please, provide your socket connector config file.

ZhaoLinl, 2024-06-11:
> 你好@ZhaoLinl，感谢您对 ThingsBoard IoT 网关的关注！ 请提供您的插座连接器配置文件。

{
  "name": "TCP Connector Example",
  "type": "TCP",
  "address": "127.0.0.1",
  "port": 50000,
  "bufferSize": 1024,
  "devices": [
    {
      "addressFilter": "127.0.0.1:*",
      "deviceName": "Device Example",
      "deviceType": "default",
      "encoding": "utf-8",
      "telemetry": [
        {
          "key": "temp",
          "byteFrom": 0,
          "byteTo": -1
        },
        {
          "key": "hum",
          "byteFrom": 0,
          "byteTo": 2
        }
      ],
      "attributes": [
        {
          "key": "name",
          "byteFrom": 0,
          "byteTo": -1
        },
        {
          "key": "num",
          "byteFrom": 2,
          "byteTo": 4
        }
      ],
      "attributeRequests": [
        {
          "type": "shared",
          "requestExpression": "${[0:3]==atr}",
          "attributeNameExpression": "[3:]"
        }
      ],
      "attributeUpdates": [
        {
          "encoding": "utf-16",
          "attributeOnThingsBoard": "sharedName"
        }
      ],
      "serverSideRpc": [
        {
          "methodRPC": "rpcMethod1",
          "withResponse": true,
          "methodProcessing": "write",
          "encoding": "utf-8"
        }
      ]
    }
  ]
}

closed 2024-06-11 by imbeacon