thingsboard/thingsboard-gateway #1425: Custom serial connector [BUG]

ZhaoLinl, 2024-06-12:
**Describe the bug**
Clear and concise description of what the bug is.
2024-06-12 11:59:54 - |INFO| - [tb_gateway_service.py] - tb_gateway_service - __init__ - 218 - Gateway starting...
2024-06-12 11:59:54 - |INFO| - [tb_gateway_service.py] - tb_gateway_service - __init__ - 223 - ThingsBoard IoT gateway version: 3.4.6
2024-06-12 11:59:54 - |WARNING| - [tb_updater.py] - tb_updater - check_for_new_version - 91 - Cannot connect to the update service. PLease check your internet connection.
2024-06-12 11:59:54 - |INFO| - [tb_gateway_mqtt.py] - tb_gateway_mqtt - gw_subscribe_to_attribute - 199 - Subscribed to *|* with id 1 for device *
2024-06-12 11:59:55 - |INFO| - [tb_gateway_service.py] - tb_gateway_service - __init__ - 309 - Gateway started.
2024-06-12 11:59:55 - |ERROR| - [tb_logger.py] - tb_logger - exception - 102 - Error on loading connector: TypeError('can only concatenate str (not "NoneType") to str')
Traceback (most recent call last):
  File "/home/root1/.local/lib/python3.8/site-packages/thingsboard_gateway-3.4.6-py3.8.egg/thingsboard_gateway/gateway/tb_gateway_service.py", line 788, in _load_connectors
    connector_class = TBModuleLoader.import_module(connector_type,
  File "/home/root1/.local/lib/python3.8/site-packages/thingsboard_gateway-3.4.6-py3.8.egg/thingsboard_gateway/tb_utility/tb_loader.py", line 48, in import_module
    buffered_module_name = extension_type + module_name
TypeError: can only concatenate str (not "NoneType") to str
2024-06-12 11:59:56 - |INFO| - [tb_gateway_remote_configurator.py] - tb_gateway_remote_configurator - __init__ - 65 - Remote Configurator started
2024-06-12 11:59:56 - |INFO| - [tb_gateway_remote_configurator.py] - tb_gateway_remote_configurator - process_config_request - 250 - Configuration update request received.

**Connector name (If bug in the some connector):**

**Error traceback (If available):**
```
'deviceName'
Traceback (most recent call last):
  File "<input>", line 2, in <module>
KeyError: 'deviceName'
```

**Versions (please complete the following information):**
 - OS: [e.g. Ubuntu 18.04]
 - Thingsboard IoT Gateway version [e.g. 3.4.6]
 - Python version[e.g. 3.9]

ZhaoLinl, 2024-06-12:
"connectors": [
   {
      "name": "Custom Serial Connector",
      "type": "serial",
      "configuration": "custom_serial.json"
    }
  ]

samson0v, 2024-06-14:
Hi @ZhaoLinl,

Thank you for your interest in the ThingsBoard IoT Gateway. 
To use extension - you need to set "class" parameter and assign the classname of the extension to it, for example:

```
"connectors": [
    {
    "name": "Custom Serial Connector",
    "type": "serial",
    "class": "CustomSerialConnector",
    "configuration": "custom_serial.json"
    }
]
```

ZhaoLinl, 2024-06-14:
Thank you for your solution. I'll give it a try.

closed 2024-06-19 by imbeacon

Cosmo997, 2024-11-10:
I have similar issue,  if i set the "class" parameter with the correct info next time i restart the connector (docker version 3.4.4 ) the "class" line inside the tb_gateway.json disappears and give me the error.