thingsboard/thingsboard-gateway #1431: [HELP]  Message publish failed: The client is not currently connected.

DuXiaoChuang, 2024-06-21:
**Describe the issue**
2024-06-21 14:17:09 - |ERROR| - [tb_logger.py] - tb_logger - exception - 98 - Message publish failed: The client is not currently connected.
Traceback (most recent call last):
  File "C:\thingsboard-gateway-master\thingsboard-gateway-master\thingsboard_gateway\gateway\tb_gateway_service.py", line 1205, in __read_data_from_storage
    else:
  File "C:\thingsboard-gateway-master\thingsboard-gateway-master\venv\Lib\site-packages\tb_device_mqtt.py", line 179, in get
    self.message_info.wait_for_publish(timeout=1)
  File "C:\thingsboard-gateway-master\thingsboard-gateway-master\venv\Lib\site-packages\paho\mqtt\client.py", line 362, in wait_for_publish
    raise RuntimeError('Message publish failed: %s' % (error_string(self.rc)))
RuntimeError: Message publish failed: The client is not currently connected.

imbeacon, 2024-06-25:
Hi, please try to use a version from the master branch, I have added handling, for this error. It should resend failed message.

DuXiaoChuang, 2024-06-25:
> Hi, please try to use a version from the master branch, I have added handling, for this error. It should resend failed message.

> Hi, please try to use a version from the master branch, I have added handling, for this error. It should resend failed message.

Hello, can you tell me which code fixes this bug?

imbeacon, 2024-06-25:
Yes, in this part of code we should handle all errors and just retry the message sending.

DuXiaoChuang, 2024-06-25:
> Yes, in this part of code we should handle all errors and just retry the message sending.

> Yes, in this part of code we should handle all errors and just retry the message sending.

Okay, let me pull the latest code and verify it. Thank you very much.

closed 2024-11-06 by imbeacon