thingsboard/thingsboard-gateway #1434: Need help understanding more about LoRa

kookohd, 2024-06-24:
**Describe the issue**
I was wondering if it is possible to connect LILYGO® T3S3 V1.0 ESP32-S3 LoRa SX1280 2.4G Development Board to rak gateway and subsequently to my chirpstack and thingsboard server as my rak gateway is used as a packet forwarder.

**Versions (please complete the following information):**
-windows 11
-chirpstack v4
-rak7268
-thingsboard v3.6.3

kookohd, 2024-06-24:
@samson0v and @imbeacon , sorry for the ping, but if you guys can help me out it will really help me out. Are you guys familiar with this?

imbeacon, 2024-06-24:
Hi @kookohd,

You can do this, but not via the gateway, you need to use integrations (They are available on PaaS or PE versions of ThingsBoard) to connect to network server. For example, the following device connects through network server as well as you want to do - [guide](https://thingsboard.io/docs/pe/devices-library/ws202-pir-and-light-sensor/).

kookohd, 2024-06-24:
Thanks for the speedy reply, the device that I am creating is like a Arduino project, it doesn't have any connectivity to smart phone, have you done something like using lora library on arduino to generate dev eui and app key before?
I am using chirpstack server and thingsboard server commmunity version. I have gathered some experience since the last time, like the experience on how to connect devices with dev eui and app key to chirpstack server and I have also been able to integrate chirpstack to thingsboard community version which is free.

So right now I am going to connect the devices using lora library will let you guys know my progress this week. I am trying to use the library to see if i can generate the dev eui and app key and I will also look further into lorawan github repository

imbeacon, 2024-06-24:
Unfortunately, we don't have a guide like you need, but I think it will contain the same process to simplify connection. It is possible to connect using the gateway, but you need to configure it to work with Chirpstack API using mqtt. It is much more complex task.

closed 2024-07-01 by imbeacon