thingsboard/thingsboard-gateway #1435: [HELP]

DuXiaoChuang, 2024-06-24:
**Describe the issue**
Create description about your issue, and your actions to solve it.

**Configuration** (Attach your configuration file)
**Notate: Remove Access token from file if you want to attach a tb_gateway.yaml**

**Connector name (If you need help with some connector/converter):**
[e.g. MQTT Connector]

**Error traceback (If it was raised):**
```
'deviceName'
Traceback (most recent call last):
  File "<input>", line 2, in <module>
KeyError: 'deviceName'
```

**Versions (please complete the following information):**
 - OS: [e.g. Ubuntu 20.04]
 - Thingsboard IoT Gateway version [e.g. 3.0.1]
 - Python version[e.g. 3.9]

DuXiaoChuang, 2024-06-24:
Error screenshot
[image]

imbeacon, 2024-07-01:
Please recreate the issue and provide more information, because there it is not clear what is wrong, with this description.

closed 2024-07-01 by imbeacon