thingsboard/thingsboard-gateway #1436: [HELP] bytes_modbus_uplink_converter - convert - 84 - Exception Response(131, 3, IllegalAddress)

knityanand, 2024-06-25:
**Describe the issue**
I am trying to get the data from energy meters with the following data config
[image]
[image]

and I am getting a NULL value and then the exception mentioned below
bytes_modbus_uplink_converter - convert - 84 - Exception Response(131, 3, IllegalAddress)

I get it for both the float and swapped float. I guess, **I am doing something wrong in the modbus.json. Would appreciate some help** Also what changes need to be done for a swapped float

**Configuration** (Attach your configuration file)
[modbus.json](https://github.com/user-attachments/files/15977293/modbus.json)

**Connector name (If you need help with some connector/converter):**
modbus.json

**Error traceback (If it was raised):**
[error.log](https://github.com/user-attachments/files/15977216/error.log)

**Versions (please complete the following information):**
 - OS: Windows 11
 - Thingsboard IoT Gateway version 3.2
 - Python version 3.10.2

DuXiaoChuang, 2024-06-28:
你的地址错误了，你试试地址+1或者-1

knityanand, 2024-07-10:
@DuXiaoChuang Thanks so much.. This worked

knityanand, 2024-07-10:
@DuXiaoChuang  Thanks so much.. This worked. Wondering why I have to add / subtract 1 from the address which works in modbus tester or modscan64.

knityanand, 2024-08-07:
Closing this. @DuXiaoChuang  Thx again..

closed 2024-08-07 by knityanand

DuXiaoChuang, 2024-08-08:
> Closing this. @DuXiaoChuang Thx again..

You're welcome. Can I add you to wechat? Continue to communicate if you have any questions.