thingsboard/thingsboard-gateway #1439: [BUG]  Write method doesn't allow!

zlwjzf, 2024-06-26:
**Describe the bug**
After kepware is connected to an Omron device and the opcua configuration of kepware is enabled, the tb-gateway is connected through the opcua protocol. Point values can be collected but cannot be modified. (The test point has read and write permissions)

opcua.json：

`{
  "server": {
    "name": "OPC-UA-Test1",
    "url": "42.64.26.253:49320",
    "timeoutInMillis": 5000,
    "scanPeriodInMillis": 5000,
    "disableSubscriptions": false,
    "subCheckPeriodInMillis": 100,
    "showMap": true,
    "security": "Basic128Rsa15",
    "identity": {
      "username": "username",
      "password": "password"
    },
    "mapping": [
      {
        "deviceNodePattern": "${ns=2;s=SM1.GJYB}",
        "deviceNamePattern": "GJYB",
        "deviceTypePattern": "GJYB",
        "attributes": [],
        "timeseries": [
          {
            "key": "Z1_ZA1_ONI",
            "path": "${ns=2;s=SM1.GJYB.Z1_ZA1_ONI}"
          },
          {
            "key": "Z1_ZA1_ONO",
            "path": "${ns=2;s=SM1.GJYB.Z1_ZA1_ONO}"
          }
        ]
      }
    ]
  }
}`

tb-gateway log：
` - |DEBUG| - [opcua_connector.py] - opcua_connector - datachange_notification - 661 - [SUBSCRIPTION] Data to ThingsBoard: {'deviceName': 'GJYB', 'deviceType': 'GJYB', 'attributes': [], 'telemetry': [{'ts': 1719391815118, 'values': {'Z1_ZA1_ONO': 'False'}}]}"
- |DEBUG| - [opcua_connector.py] - opcua_connector - __search_node - 560 - Looking for node with config"
- |DEBUG| - [opcua_connector.py] - opcua_connector - __search_node - 565 - Found in ns=2;s=SM1.GJYB.Z1_ZA1_ONO"
- |ERROR| - [opcua_connector.py] - opcua_connector - server_side_rpc_handler - 303 - Write method doesn't allow!"`

operation procedure：
Calls the rpc-v-2-controller interface in swagger-ui with the following parameters:
`{
  "method": "set",
  "params": 
    "ns=2;s=SM1.GJYB.Z1_ZA1_ONO;value=true"
  ,
  "persistent": true,
  "timeout": 5000
}`

**Versions (please complete the following information):**
 - OS: [windows7 docker]
 - Thingsboard IoT Gateway version [3.1]
 - Python version[e.g. 3.7]

zlwjzf, 2024-07-10:
Hello, I have found the cause of the problem. When calling the RPC set method, replace node.set_value(value) in the source opcua_connector.py file with the following code to achieve the simulated opcua point in kepware to write values
`dv = ua.DataValue(ua.Variant(int(value), ua.VariantType.Int16))`
`dv = ua.DataValue(value)`
`dv.ServerTimestamp = None`
`dv.SourceTimestamp = None`
`node.set_value(dv)`
Can you adapt these changes to the latest version?

zlwjzf, 2024-07-10:
https://github.com/FreeOpcUa/opcua-asyncio/issues/30

shanjunying, 2024-08-23:
I also encountered the same problem, writing to the opcua service provided by kepserver resulted in the error 
`"opcua. ua. uaerrors. auto. BadWriteNotSupported:" The server does not support writing the combination of value, status, and timestamps provided. "(BadWriteNotSupported)`
 but writing to the service provided by Prosys OPC UA Simulation Server did not cause any problems

shanjunying, 2024-08-23:
> `dv = ua.DataValue(ua.Variant(int(value), ua.VariantType.Int16))`
> `dv.ServerTimestamp = None`
> `dv.SourceTimestamp = None`
> `node.set_value(dv)`

Where should these codes be added？Please tell me, thank you！

zlwjzf, 2024-09-03:
thingsboard_gateway/connectors/opcua/opcua_connector.py
In method `server_side_rpc_handler`
`if rpc_method == 'get':`
`  self.__gateway.send_rpc_reply(content['device'],`
`  content['data']['id'],`
`  {content['data']['method']: node.get_value(),`
`  'code': 200})`
`else:`
`......`
Load in else

closed 2025-05-08 by imbeacon