thingsboard/thingsboard-gateway #1440: [HELP]

Isac-enerlab, 2024-06-26:
help (error has occurred: 'RemoteStation' object has no attribute 'addrTuple')

I have several gateways collecting data, but we are encountering the problem "Rate limits exceeded. Rate limits for transport messages per device exceeded." These alerts are occurring because the ThingsBoard platform is receiving too many messages to process. We believe this is primarily indicated by the log below: AttributeError: 'RemoteStation' object has no attribute 'addrTuple'.

I made some attempts to modify the script to limit the number of logs, but when this change was made, I lost the data collection functionality, so I reverted to the default.

Here are the logs:

Jun 26 13:05:39 GW-SCP-1 thingsboard-gateway[9651]:   File "/home/enerlab-1/.local/lib/python3.11/site-packages/bacpypes/appservice.py", line 1471, in indication
Jun 26 13:05:39 GW-SCP-1 thingsboard-gateway[9651]:     self.sap_request(xpdu)
Jun 26 13:05:39 GW-SCP-1 thingsboard-gateway[9651]:   File "/home/enerlab-1/.local/lib/python3.11/site-packages/bacpypes/comm.py", line 527, in sap_request
Jun 26 13:05:39 GW-SCP-1 thingsboard-gateway[9651]:     self.serviceElement.indication(*args, **kwargs)
Jun 26 13:05:39 GW-SCP-1 thingsboard-gateway[9651]:   File "/home/enerlab-1/.local/lib/python3.11/site-packages/thingsboard_gateway/connectors/bacnet/bacnet_utilities/tb_gateway_bacnet_application.py", line 70, in indication
Jun 26 13:05:39 GW-SCP-1 thingsboard-gateway[9651]:     apdu.pduSource.addrTuple[0],
Jun 26 13:05:39 GW-SCP-1 thingsboard-gateway[9651]:     ^^^^^^^^^^^^^^^^^^^^^^^^
Jun 26 13:05:39 GW-SCP-1 thingsboard-gateway[9651]: AttributeError: 'RemoteStation' object has no attribute 'addrTuple'
Jun 26 13:06:21 GW-SCP-1 thingsboard-gateway[9651]: 2024-06-26 13:06:21 - |ERROR| - [core.py] - core - run - 174 - an error has occurred: 'RemoteStation' object has no attribute 'addrTuple'
Jun 26 13:06:21 GW-SCP-1 thingsboard-gateway[9651]: Traceback (most recent call last):
Jun 26 13:06:21 GW-SCP-1 thingsboard-gateway[9651]:   File "/home/enerlab-1/.local/lib/python3.11/site-packages/bacpypes/core.py", line 165, in run

This problem occurs similarly in two systems that we use because we are migrating from Debian to Ubuntu:

Operating System: Linux Debian 12.1
Hardware: Pi4 Model B
Memory Usage: 338 / 3793 MB (8.9%)
Disk Usage: 8108 / 29392 MB (29.1%)
Python 3.11.2
TBGateway{\"current_version\": \"3.5.1\", \"latest_version\": \"3.5.1\"}

The other system we use, which is the primary for us:

Operating System: Linux Ubuntu 22.04 Jammy
Hardware: Pi4 Model B
Memory Usage: 404 / 3790 MB (10.7%)
Disk Usage: 4385 / 59659 MB (7.7%)
Python 3.10.12
TBGateway TBGateway{\"current_version\": \"3.5.1\", \"latest_version\": \"3.5.1\"}

I followed the procedures indicated in this channel about a year ago but without success. 
https://github.com/thingsboard/thingsboard-gateway/issues/1044

The only remaining procedure is a complete reinstallation from scratch. However, I am limited in making this replacement because 4 units with this problem are installed at the client's site. I have 14 more units installed without this issue.

I intend to expand, but I need to have this "Rate limits exceeded. Rate limits for transport messages per device exceeded." under control.

Can you help me?

Isac-enerlab, 2024-07-03:
Still waiting for a response please!

I have new data, same situation on another rpi installed on one of the units where we collect data.

Operating System: Linux Ubuntu 22.04 Jammy
Hardware: Pi4 Model B
Memory Usage: 474 / 3790 MB  
Disk Usage: 4385 / 59659 MB 
Python 3.10.12
TBGateway TBGateway{"current_version": "3.5", "latest_version": "3.5.1"}

python3[1755407]:     self.serviceElement.indication(*args, **kwargs)
python3[1755407]:   File "/usr/lib/python3/dist-packages/thingsboard_gateway/connectors/bacnet/bacnet_utilities/tb_gateway_bacnet_application.py", line 70, in indication
python3[1755407]:     apdu.pduSource.addrTuple[0],
python3[1755407]: AttributeError: 'RemoteStation' object has no attribute 'addrTuple'

Some attempts were made to add a check for the "RemoteStation" object before trying to access the "addrTuple" attribute.

But these are corrections that are of no use on a large scale, we have 18 units monitored and we cannot make corrections unit by unit. We need to identify and correct the problem as we have an expansion plan.

Please can you guide me

Isac-enerlab, 2024-07-04:
Some more relevant data.

Granting the correct permissions to the directory that stores the telemetry in “sqlite” storage, the service works by performing the storage, however errors related to:

AttributeError: 'RemoteStation' object has no attribute 'addrTuple'

They remain, increasing the use of “Scripts hourly activity" in ”api-usage”.

Please

samson0v, 2024-12-03:
Hi @Isac-enerlab and everyone!
We are pleased to inform you that we added a new Async BACnet connector that used bacpypes3 (via PR #1607).
The old BACnet connector is now deprecated and can only be enabled manually.
Please, run the latest version of the gateway via the master branch and let us know about the results.

imbeacon, 2024-12-11:
This issue closed, because relates to outdated implementation of connector, please try the latest version (3.6.2), and feel free to open a new issue if you discover some misbehaviour.

closed 2024-12-11 by imbeacon