thingsboard/thingsboard-gateway #1441: [BUG] 2024-06-27 10:29:48 - |ERROR| - [tb_device_mqtt.py] - tb_device_mqtt - _publish_data - 775 - Timeout while waiting for a publish to ThingsBoard!

DuXiaoChuang, 2024-06-27:
[image]

imbeacon, 2024-07-01:
Hi @DuXiaoChuang,

Do you send to any of ThingsBoard clouds? demo.thingsboard.io/thingsboard.cloud? If yes, please check - do you exceed the rate limit or no and let us know. You can do this by checking notifications.

DuXiaoChuang, 2024-07-01:
> Hi @DuXiaoChuang,
> 
> Do you send to any of ThingsBoard clouds? demo.thingsboard.io/thingsboard.cloud? If yes, please check - do you exceed the rate limit or no and let us know. You can do this by checking notifications.

Yes, I'm pushing data to thingsboard's cloud platform. Are there any limits on how much data the cloud platform can push?

imbeacon, 2024-07-01:
Yes, there are strict rate limits on thingsboard.cloud, you may find information about them [here](https://thingsboard.io/products/paas/subscription/). 
The latest python client sdk waits until they will be cleared and only then send messages, but timeouts still can appear, if you have put a lot of messages for sending.

DuXiaoChuang, 2024-07-02:
> Yes, there are strict rate limits on thingsboard.cloud, you may find information about them [here](https://thingsboard.io/products/paas/subscription/). The latest python client sdk waits until they [...]

ok，thanks

closed 2024-11-06 by imbeacon