thingsboard/thingsboard-gateway #1442: [BUG] v3.5.1 crash immediatelly when start.

fjpanag, 2024-06-27:
**Describe the bug**

I just tried v3.5.1, but it fails to start, with an exception right at the start:

```
smartmaic-gw  | 2024-06-27 15:43:15 - |INFO| - [tb_gateway_service.py] - tb_gateway_service - __init__ - 225 - Gateway starting...
smartmaic-gw  | 2024-06-27 15:43:15 - |INFO| - [tb_gateway_service.py] - tb_gateway_service - __init__ - 228 - ThingsBoard IoT gateway version: 3.5.1
smartmaic-gw  | 2024-06-27 15:43:15 - |INFO| - [tb_client.py] - tb_client - _check_certificates - 198 - Will generate new certificate
smartmaic-gw  | /thingsboard_gateway/tb_utility/tb_utility.py:233: CryptographyDeprecationWarning: Properties that return a naïve datetime object have been deprecated. Please switch to not_valid_after_utc.
smartmaic-gw  |   if cert_detail.not_valid_after - datetime.datetime.now() <= datetime.timedelta(days=days_left):
smartmaic-gw  | Traceback (most recent call last):
smartmaic-gw  |   File "/usr/local/lib/python3.11/logging/__init__.py", line 1742, in isEnabledFor
smartmaic-gw  |     return self._cache[level]
smartmaic-gw  |            ~~~~~~~~~~~^^^^^^^
smartmaic-gw  | KeyError: 'DEBUG'
smartmaic-gw  | 
smartmaic-gw  | During handling of the above exception, another exception occurred:
smartmaic-gw  | 
smartmaic-gw  | Traceback (most recent call last):
smartmaic-gw  |   File "/thingsboard_gateway/tb_gateway.py", line 41, in <module>
smartmaic-gw  |     main()
smartmaic-gw  |   File "/thingsboard_gateway/tb_gateway.py", line 33, in main
smartmaic-gw  |     TBGatewayService(path.dirname(path.abspath(__file__)) + '/config/tb_gateway.json'.replace('/', path.sep))
smartmaic-gw  |   File "/thingsboard_gateway/gateway/tb_gateway_service.py", line 231, in __init__
smartmaic-gw  |     self.tb_client = TBClient(self.__config["thingsboard"], self._config_dir, connection_logger)
smartmaic-gw  |                      ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
smartmaic-gw  |   File "/thingsboard_gateway/gateway/tb_client.py", line 64, in __init__
smartmaic-gw  |     self._create_mqtt_client(config['security'])
smartmaic-gw  |   File "/thingsboard_gateway/gateway/tb_client.py", line 174, in _create_mqtt_client
smartmaic-gw  |     if self.__logger.isEnabledFor('DEBUG'):
smartmaic-gw  |        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
smartmaic-gw  |   File "/usr/local/lib/python3.11/logging/__init__.py", line 1746, in isEnabledFor
smartmaic-gw  |     if self.manager.disable >= level:
smartmaic-gw  |        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
smartmaic-gw  | TypeError: '>=' not supported between instances of 'int' and 'str'
```

**Connector name (If bug in the some connector):**
I am using the MQTT connector, but it seems unrelated.

**Error traceback (If available):**
*See above*

**Versions (please complete the following information):**
 - Official Docker image
 - Thingsboard IoT Gateway version: v3.5.1

imbeacon, 2024-07-01:
Hi @fjpanag,

Thank you for your interest in ThingsBoard IoT Gateway. Please try the version from the master branch, this bug should be solved there.

closed 2024-07-01 by imbeacon

AndreKR, 2024-08-20:
Will there be a release soon? The fix is in master for a while now but it hasn't made it into the `latest` Docker image yet.