thingsboard/thingsboard-gateway #1446: [BUG] Storage: Database is locked

elgutierrez, 2024-06-29:
**Describe the bug**
I'm using a pretty default config, with a SQLite storage configuration.
```
"storage": {
    "type": "sqlite",
    "read_records_count": 100,
    "max_records_count": 100000,
    "data_folder_path": "./data/",
    "max_file_count": 10,
    "max_read_records_count": 10,
    "max_records_per_file": 10000,
    "data_file_path": "/etc/thingsboard-gateway/data/data.db",
    "messages_ttl_check_in_hours": 1,
    "messages_ttl_in_days": 7,
    "ts": 1717605380034
  },
  ```
It works OK for a while, but sometimes I start getting this error:
```
python3[1034]: 2024-06-29 07:26:47 - |ERROR| - [database_connector.py] - database_connector - execute - 66 - database is locked
python3[1034]: Traceback (most recent call last):
python3[1034]:   File "/usr/lib/python3/dist-packages/thingsboard_gateway/storage/sqlite/database_connector.py", line 62, in execute
python3[1034]:     return self.connection.execute(*args)
python3[1034]: sqlite3.OperationalError: database is locked
```
And what's worse is that the devices that display this behavior start to loop on it in a high frequency. I suspect that it then tries to report the error to the TB server for incrementing the error counter our something like this. But as the frequency is super high, I start to get 10x data volume, eating out my API quota and triggering Rate Limits.

**Connector name:**
Database connector

**Error traceback:**
```
python3[1034]: Traceback (most recent call last):
python3[1034]:   File "/usr/lib/python3/dist-packages/thingsboard_gateway/storage/sqlite/database_connector.py", line 62, in execute
python3[1034]:     return self.connection.execute(*args)
python3[1034]: sqlite3.OperationalError: database is locked
```

**Versions:**
 - OS: Ubuntu 22.04.4 LTS
 - Thingsboard IoT Gateway version 3.4.5
 - Python version: 3.7

samson0v, 2024-07-01:
Hi @elgutierrez, thanks for your interest in ThingsBoard IoT Gateway!
Can you please provide your config files? Because for now, I can't reproduce your issue.
Also, could you try using the latest version (3.5.1) and tell us about the result?

samson0v, 2024-07-08:
@elgutierrez as I can see from the commits history, you need to use the version from the master branch.

elgutierrez, 2024-07-08:
So it's not released yet? Do you have an ETA for the release?

samson0v, 2024-07-08:
@elgutierrez on this week

elgutierrez, 2024-07-08:
On another note, I realized that having an error in the gateway (could be this issue with the DB, or maybe something else), triggers a behavior where the gateway starts looping very fast, retrying the failed command and getting the same issue. In those cases, the API is bombarded with tens/hundreds of requests per second, leading to a rate limit issue. I suspect this is caused by the way the gateway reports errors. Is there a related open issue that you know of?

imbeacon, 2024-07-10:
Hi @elgutierrez,

Yes, we know about situations like this and it relates to the implementation of base library, at the moment we are trying to find the solution for this issue. We can just suggest to disable remote logging, to avoid a lot of messages.
At the same time please describe when did you encounter the issue?

elgutierrez, 2024-07-18:
Remote logging is already disabled, it was the 1st thing I did. I guess it's more likely to be related with the error counters like
`Bacnet_ERRORS_COUNT`, `storage_ERRORS_COUNT`, etc

imbeacon, 2024-07-30:
I have adjusted the logger, to send errors count every 5 seconds instead of on each error message. You can try to use a version from the master branch.

samson0v, 2024-08-06:
Hi @elgutierrez, any updates?

elgutierrez, 2024-08-19:
Hey, unfortunately I can't update my field devices with code from master branch, so I'm waiting for the new release so I can update it. Do we have an updated ETA?

closed 2024-09-16 by imbeacon