thingsboard/thingsboard-gateway #1449: I have modified the code on this project, may I ask how to package the latest code and run it without IDE?

DuXiaoChuang, 2024-07-03:
I have modified the code on this project, may I ask how to package the latest code and run it without IDE? What does the for_bulid folder for this picture do

[image]

imbeacon, 2024-07-03:
Hi @DuXiaoChuang,

It depends on what kind of distribution you want, if you want just python package you can run `python3 setup.py sdist` or `python3 -m build` in the root folder. In case you need a debian or rpm package you will need to run make_packages.sh script. If you want to create your own debian image - you can copy Dockerfile from docker folder to the root and just run `docker build . -t my-gateway` (replace my-gateway with any tag that you want).

DuXiaoChuang, 2024-07-03:
> Hi @DuXiaoChuang,
> 
> It depends on what kind of distribution you want, if you want just python package you can run `python3 setup.py sdist` or `python3 -m build` in the root folder, the package will be in `dist` folder. In case you need a debian or rpm package you will need to run make_packages.sh script. If you want to create your own debian image - you can copy Dockerfile from docker folder to the root and just run `docker build . -t my-gateway` (replace my-gateway with any tag that you want).

error message is displayed after run make packages.sh script
[image]

DuXiaoChuang, 2024-07-03:
> Hi @DuXiaoChuang,
>
> It depends on what kind of distribution you want, if you want just python package you can run `python3 setup.py sdist` or `python3 -m build` in the root folder, the package will [...]

`python3 setup.py sdist` or `python3 -m build`

imbeacon, 2024-07-03:
`python3 setup.py sdist` creates python source distribution and you can install it using the following `pip install dist/thingsboard-gateway-3.5.1.tar.gz`

According to the issue with make_packages - you need to install additional packages, using command `pip3 install bdist_rpm`, as far as I remember, please try the first option with pip and if it won’t help you, I will check libraries that required to build rpm package.

DuXiaoChuang, 2024-07-03:
> `python3 setup.py sdist` creates python source distribution and you can install it using the following `pip install dist/thingsboard-gateway-3.5.1.tar.gz`
>
> According to the issue with make_packages [...]

I successfully installed /thingsboard-gateway-3.4.6.tar.gz on windows. How do I start and run

imbeacon, 2024-07-03:
You can run it for example using the following command: `python3 path/to/installed/package/thingsbaord_gateway/tb_gateway.py`

closed 2024-11-06 by imbeacon