thingsboard/thingsboard-gateway #1450: [HELP] Connectors RPC on UI setValue getValue dont work???

Jonnik84, 2024-07-03:
Hi there 
Im trying to understand is it possible to read any register or write any register manually from TB UI using  conector RPC dashboard screen (attaching)
[image]

But when i try to setValue or getValue, errors in log appear:
[image]

Pls help how to use this   or maybe its not working yet??

tatuiot, 2024-07-04:
Hi guys, I am facing the same issue here, but with mqtt connector. The good thing is I can send my attributes and telemetry from device -> mqtt -> gateway -> TB server.  no issues with that I can dynamically create devices, which is GREAT. but I cannot get RPC working. I created a dumb dashboard with a round switch to set pin value to my device. I am stuck for over a week now... no luck so far, I created a whole new environmet, with python installation for a new gateway... same issue, 

here is my whole mqtt.json file just in case...  but I cannot get any attribute or RPC update on the device side ... thanks a lot for your help. 

My config is the following
Gateway ver 3.5.1 - Raspberry Pi installed using the python method. 
TB Server CE ver 3.6.4. Ubuntu 20.04 . running 
Device for testing also the RPi simulating mqtt devices.

as a 3rd issue...  I have this message in the LOGS Gateway dashboard table 

Logging loading exception, logs.json is wrong: Unable to configure handler 'connectorHandler'

```

{
 "broker": {
    "name": "Default Local Broker",
    "host": "127.0.0.1",
    "port": 1883,
    "clientId": "ThingsBoard_gateway",
    "version": 5,
    "maxMessageNumberPerWorker": 10,
    "maxNumberOfWorkers": 100,
    "sendDataOnlyOnChange": false,
    "security": {
      "type": "basic",
      "username": "tronador",
      "password": "a3m8zljj"
    }
  },
  "mapping": [
    {
      "topicFilter": "sensor/data",
      "converter": {
        "type": "json",
        "deviceNameJsonExpression": "${serialNumber}",
        "deviceTypeJsonExpression": "${sensorType}",
        "sendDataOnlyOnChange": false,
        "timeout": 60000,
        "attributes": [
          {
            "type": "string",
            "key": "model",
            "value": "${sensorModel}"
          },
          {
            "type": "bool",
            "key": "pin0",
            "value": "${pin0}"
          },
          {
            "type": "bool",
            "key": "pin1",
            "value": "${pin1}"
          }
        ],
"timeseries": [
          {
            "type": "double",
            "key": "temperature",
            "value": "${temp}"
          },
          {
            "type": "double",
            "key": "humidity",
            "value": "${hum}"
          },
          {
            "type": "double",
[... 127 lines cut ...]
      "deviceNameFilter": ".*",
      "attributeFilter": "model",
      "topicExpression": "/sensor/${deviceName}/${attributeKey}",
      "valueExpression": "{\"${attributeKey}\":\"${attributeValue}\"}"
    }
  ],
  "serverSideRpc": [
    {
      "deviceNameFilter": ".*",
      "methodFilter": "set",
      "requestTopicExpression": "/sensor/${deviceName}/request/${methodName}/${requestId}",
      "responseTopicExpression": "/sensor/${deviceName}/response/${methodName}/${requestId}",
      "responseTimeout": 10000,
      "valueExpression": "${params}"
    },
    {
      "deviceNameFilter": ".*",
      "methodFilter": "no-reply",
      "requestTopicExpression": "/sensor/${deviceName}/request/${methodName}/${requestId}",
      "valueExpression": "${params}"
    }
  ],
  "id": "a209bf2b-f196-49a7-872f-d3877fbc5cc4"
}
```

samson0v, 2024-07-11:
Hi @Jonnik84 and @tatuiot, thanks for your interest in the ThingsBoard IoT Gateway!
We know about this bug and it was fixed, so you can use the version from the master branch or wait for the next release (11.07.2024).

Jonnik84, 2024-07-12:
Thanks a lot seems to work now

closed 2024-10-14 by imbeacon