thingsboard/thingsboard-gateway #1454: [HELP] Install ThingsBoard IoT Gateway as package-arm64-ubuntu-server

pro100805, 2024-07-09:
during installation the following message appears (version conflict)

[Install ThingsBoard IoT Gateway as package-arm64-ubuntu-server.txt](https://github.com/user-attachments/files/16149608/Install.ThingsBoard.IoT.Gateway.as.package-arm64-ubuntu-server.txt)

pro100805, 2024-07-13:
[Install ThingsBoard IoT Gateway as package-ubuntu-server-24.04.txt](https://github.com/user-attachments/files/16201262/Install.ThingsBoard.IoT.Gateway.as.package-ubuntu-server-24.04.txt)

imbeacon, 2025-04-17:
Hi @pro100805,

In the latest version (3.7.3) of the gateway we creating a separated virtual environment for it during installation from debian package and this issue is no longer reproducible, because all packages will be installed into created virtual environment.