thingsboard/thingsboard-gateway #1460: [HELP]gateway should mix devices and connections 1toM and Mto1

Jonnik84, 2024-07-13:
Hi.
I’m attaching a small diagram to help to understand my question.

In my opinion,  Gateway is a device that should transform data from one system to another system and that’s why there should be no direct connection between connectors and cloud devices.

For example, I have one modbus IO module, to which I connect several temperature sensors. For now there is no problem to create several modbus connectors each four different device in the cloud,  aimed to get different registers from this IOModule. Gateway UI dashboard, I will see those devices but they connect her will be the same for all of them.
[image]

but there is a problem when I have several IO modules, that I need to maintain into one cloud device 
in this case, system doesn’t work. device cannot be displayed correctly ( it collects data to one device from different connectors.) UI doesn’t simply have such functionality. It’s possible now to display only one connector for device….

For me, it’s mission critical for Gateway to have such ability. 
[gateway.pdf](https://github.com/user-attachments/files/16203130/gateway.pdf)

Jonnik84, 2024-07-30:
@imbeacon thanks for suggestion 
But I don't get. please explain.  :)))

So I need to create several cloud devices for each connector and send telemetry between them with rule chains??

imbeacon, 2024-07-30:
Particularly you are right.
When data arrives to ThingsBoard, you have a post telemetry message in rule chains, with originator - device, configured in the connector configuration.
I have created just an example rule chain, you can play with it to understand on how to duplicate message to another device, new device will be created automatically in this rule chain. It is just an example, but not a solution for yout case, please adjust it for your needs.
[test_message_splitting.json](https://github.com/user-attachments/files/16423884/test_message_splitting.json)

Example on how to connect it in root rule chain is the following, don’t forget to check is device name in metadata (yellow switch in right side on screenshot) is equal to “base” device.

[image]

closed 2024-10-14 by imbeacon