thingsboard/thingsboard-gateway #1461: can't add "CUSTOM" connector by gateway dashboard

langlichong, 2024-07-16:
**Describe the bug**
just open the gateway dashboard and add a connector , the conenctor type use "custom" provided by ui , and then click the save button ,then there is no response , the chrome console show an error 
[image]
[image]

**Versions (please complete the following information):**
 - OS: ubuntu22.04
 - Thingsboard Version: 3.7-release
 - Thingsboard IoT Gateway version 3.5.1
 - Python version 3.8.10

imbeacon, 2024-07-30:
Hi @langlichong,

Thank you for your interest in ThingsBoard IoT gateway.
Unfortunately, right now we have an issue with custom connectors and converters, assigned from UI on ThingsBoard IoT Gateways dashboard, we will solve it in the future release of ThingsBoard, and for now it is unfortunaly impossible to correctly configure custom connectors.

ivanlr, 2024-10-09:
Hi @imbeacon, from Horse we are very interested in the follow up of this issue. Do you have any update regarding this?