thingsboard/thingsboard-gateway #1469: Unable to run tb_gateway.py

rock-007, 2024-07-18:
**Describe the bug**
I have been following this guid(https://thingsboard.io/docs/iot-gateway/install/source-installation/) to run the GW locally, I have successfully run the backend springboot application and when running this command it is having exception in the thread, is it some know bug or if some settings I need to teak for it to work?
 
**Connector name (If bug in the some connector):**
not sure but it also say can't connect to GRPC connector!

**Error traceback (If available):**
```
`
'deviceName'
Traceback (most recent call last):
  ❯ python3 ./thingsboard_gateway/tb_gateway.py
Cannot load GRPC connector!
2024-07-18 14:57:19 - |INFO| - [tb_gateway_service.py] - tb_gateway_service - __init__ - 168 - Gateway starting...
2024-07-18 14:57:19 - |INFO| - [tb_gateway_service.py] - tb_gateway_service - __init__ - 171 - ThingsBoard IoT gateway version: 3.5.1
Exception in thread Thread-4 (_thread_main):
Traceback (most recent call last):
  File "/Library/Frameworks/Python.framework/Versions/3.11/lib/python3.11/threading.py", line 1045, in _bootstrap_inner
    self.run()
  File "/Library/Frameworks/Python.framework/Versions/3.11/lib/python3.11/threading.py", line 982, in run
    self._target(*self._args, **self._kwargs)
  File "/Users/umairashraf/Projects/Personal/thingsboard-gateway/venv/lib/python3.11/site-packages/paho/mqtt/client.py", line 3637, in _thread_main
    self.loop_forever(retry_first_connection=True)
  File "/Users/umairashraf/Projects/Personal/thingsboard-gateway/venv/lib/python3.11/site-packages/paho/mqtt/client.py", line 1769, in loop_forever
    rc = self._loop(timeout)
         ^^^^^^^^^^^^^^^^^^^
  File "/Users/umairashraf/Projects/Personal/thingsboard-gateway/venv/lib/python3.11/site-packages/paho/mqtt/client.py", line 1177, in _loop
    rc = self.loop_read()
         ^^^^^^^^^^^^^^^^
  File "/Users/umairashraf/Projects/Personal/thingsboard-gateway/venv/lib/python3.11/site-packages/paho/mqtt/client.py", line 1569, in loop_read
    rc = self._packet_read()
         ^^^^^^^^^^^^^^^^^^^
  File "/Users/umairashraf/Projects/Personal/thingsboard-gateway/venv/lib/python3.11/site-packages/paho/mqtt/client.py", line 2485, in _packet_read
    rc = self._packet_handle()
         ^^^^^^^^^^^^^^^^^^^^^
  File "/Users/umairashraf/Projects/Personal/thingsboard-gateway/venv/lib/python3.11/site-packages/paho/mqtt/client.py", line 3075, in _packet_handle
    return self._handle_pubackcomp("PUBACK")
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/umairashraf/Projects/Personal/thingsboard-gateway/venv/lib/python3.11/site-packages/paho/mqtt/client.py", line 3565, in _handle_pubackcomp
    reasonCode.unpack(self._in_packet['packet'][2:])
  File "/Users/umairashraf/Projects/Personal/thingsboard-gateway/venv/lib/python3.11/site-packages/paho/mqtt/reasoncodes.py", line 167, in unpack
    name = self.__getName__(self.packetType, c)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/umairashraf/Projects/Personal/thingsboard-gateway/venv/lib/python3.11/site-packages/paho/mqtt/reasoncodes.py", line 138, in __getName__
    assert identifier in self.names.keys(), identifier
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
AssertionError: 47
2024-07-18 14:58:20 - |WARNING| - [tb_device_mqtt.py] - tb_device_mqtt - _publish_data - 745 - Waiting for connection to be established before sending data to ThingsBoard!
2024-07-18 14:58:30 - |WARNING| - [tb_device_mqtt.py] - tb_device_mqtt - _publish_data - 745 - Waiting for connection to be established before sending data to ThingsBoard!
`
KeyError: 'Exception in thread Thread-4 (_thread_main):'
```

**Versions (please complete the following information):**
 - OS: Mac
 - Thingsboard IoT Gateway version 3.5.1
 - Python version 3.11.9

rock-007, 2024-07-18:
so the above issue is resolved as the port number was the issue in the config file.
Now I am facing another issue as air complaining about the user/password is wrong, I have add the access token in the 
`thingsboard_gateway/config/tb_gateway.json` 

Note: Just to double check I  have used the user/password of `tenant@thingsboard.org/tenant` & `* sysadmin@thingsboard.org/sysadmin `, and got the access token using the browser network tab...

[image]

rock-007, 2024-07-19:
Thanks @samson0v for the reply, yes I figured out last night about this step its in the general configuration/security section, Its working now. Thanks :)

closed 2024-07-19 by rock-007

lolPlatinumPlayer, 2024-08-14:
> so the above issue is resolved as the port number was the issue in the config file. Now I am facing another issue as its complaining about the user/password is wrong, I have add the access token in the `thingsboard_gateway/config/tb_gateway.json`
> 
> Note: Just to double check I have used the user/password of `tenant@thingsboard.org/tenant` & `sysadmin@thingsboard.org/sysadmin `, and got the access token using the browser network tab...
> 
> <img alt="image" width="1157" src="https://private-user-images.githubusercontent.com/59394025/350082029-21a0f13e-3a8c-49c1-b79e-bb1877fc8b69.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjM0NzMwMzcsIm5iZiI6MTcyMzQ3MjczNywicGF0aCI6Ii81OTM5NDAyNS8zNTAwODIwMjktMjFhMGYxM2UtM2E4Yy00OWMxLWI3OWUtYmIxODc3ZmM4YjY5LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA4MTIlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwODEyVDE0MjUzN1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTE5NTQwNzc2YjlmZDRiZjQzN2YyYmU0ZWEzZGQ1ZWViNjhhNWJjNzI2M2Y4NDVlODZmMTlhMWRjYjEyNDlmY2YmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.tzIIB1WqN7GW0CzBr3uhaauQ3Kn9GDWgwpG8qRAXPZA">

Hi, @rock-007 . I see `Cannot load GRPC connector!` in this Picture. Is it means its OK?