thingsboard/thingsboard-gateway #1470: iot gateway cannot connect to thingsboard (as the log says) [HELP]

stevechen2013, 2024-07-19:
**Describe the issue**
thingsboard CE version locally installed on a ubuntu os
thingsboard iot gateway locally installed via docker+linux per instructions : https://thingsboard.io/docs/iot-gateway/install/docker-installation/

**Configuration** (Attach your configuration file)
as default
**Notate: Remove Access token from file if you want to attach a tb_gateway.yaml**

**Connector name (If you need help with some connector/converter):**
no connector

**Error traceback (If it was raised):**
```
[tb_client.py] tb_client connect 288 connection refused, check thingsboard is running
```
while on the thingsboard ui at localhost:8080, the iot gateway is active shown in the dashboard

**Versions (please complete the following information):**
 - OS: ubuntu
 - Thingsboard IoT Gateway version [3.4]
 - Python version[3.10]

tatuiot, 2024-07-22:
Hi, there is a config setup you need to configure. 
/etc/thingsboard-gateway/config/tb_config.json you need to specify where you want the gateway to report to ... in my case the TB Server CE .
then you need to create a gateway device into the TB Server, get the credential to place it into the file mentioned above.
Once you have that... then you should create a dashboard to see if the gateway is connected. 
it is so tricky, I have spent over a month so far. I can send data and RPCs from TB Server...  let me know If you managed to install the gateway using which method, each method has its own difficulty...

tyagiaksh, 2024-09-10:
hi @tatuiot 
I am having the same issue while trying to connect gateway with thingsboard, I am using community edition thingboard and following the same approach as given in documentation but getting this error
|ERROR| - [tb_client.py] - tb_client - connect - 288 - Connection refused. Check ThingsBoard is running.
And also can you tell how many connector can I connect with one gateway in thingsboard community edition

imbeacon, 2024-10-14:
Hi,

If you are running the gateway using the docker compose and ThingsBoard in the docker as well - please try to use `host.docker.internal` instead of `localhost` as host in tb_config.json. Because in case of connecting to localhost, when the gateway is in container - it tries to connect to local address inside the container.

closed 2024-11-06 by imbeacon