thingsboard/thingsboard-gateway #1476: [HELP] Unable to detect multiple devices connected to the same panel address

knityanand, 2024-07-27:
**Describe the issue**
Yabe shows two devices 

Device 4011 - 0.0.0.0:4025 via 192.168.1.12:47808

Device 4012 - 0.0.0.0:4026 via 192.168.1.12:47808

How can this be configured in bacnet.json for thingsboard iot gateway. These devices are not getting detected.

**Configuration** (Attach your configuration file)
[BACNet_2dev.json](https://github.com/user-attachments/files/16401659/BACNet_2dev.json)

**Connector name (If you need help with some connector/converter):**
BACNET Connector

**Error traceback (If it was raised):**
```

**Versions (please complete the following information):**
 - OS: Ubuntu 20.04
 - Thingsboard IoT Gateway version 3.5.1
 - Python version 3.10.2

knityanand, 2024-08-03:
Can somebody please help. This is critical for me. We just encountered this on a production env

knityanand, 2024-08-07:
I have made some progress on this... Am now able to read the data from bacpypes. Had to find the addresses as mentioned below.

wirtn *
192.168.3.12
70
192.168.3.227
50
whois : 4000 4999
4010 70:0x000000000faa
4011 70:0x000000000fab
4001 192.168.3.23
read 70:0x000000000fab binaryInput,11 presentValue
active

However, when I am use the address format as 70:0x000000000fab, nothing seems to happen in the gateway... No devices are created. Please find bacnet.json attached

[BACNet_multi.json](https://github.com/user-attachments/files/16534544/BACNet_multi.json)

Checked wireshark. 

I get responses for WhoamI but no request for read parameter seems to be going. The same works from bacpypes as shown above. What am I doing wrong?

@samson0v @imbeacon Any help will be greatly appreciated.

knityanand, 2024-08-11:
I was able to fix this issue with the following fixes. Please see below.

bacnet.json
The following was how the bacnet.json was configured
    {
      "deviceName": "Ventilation Fans Status-HVAC",
      "deviceType": "default",
      **"address": "70:0x000000000faa",**    // This was discovered using the steps in previous comment.
      "pollPeriod": 900000,
      "timeseries": [
        {
          "key": "13F_Z1_CODetector_B4_Z4_Zone-13COLeve-On/Off Status",
          "objectId": "analogInput:117",
          "propertyId": "presentValue"
        },
        {
          "key": "B1_Z1_CODetector_B1_Z1_Spare-1-On/Off Status",
          "objectId": "analogInput:134",
          "propertyId": "presentValue"
        },
        {
          "key": "B1_Z1_CODetector_B1_Z1_Spare-2-On/Off Status",
          "objectId": "analogInput:135",
          "propertyId": "presentValue"
        },

Also, for such devices, the values are not returned in addrTuple. The below log statement throws an exception and hence the callback is not registered for these devices resulting in no activity for these devices. Commenting it fixed it and also the regular devices worked.

**thingsboard_gateway\connectors\bacnet\bacnet_utilities\tb_gateway_bacnet_application.py**
    
def indication(self, apdu: APDU):
        if isinstance(apdu, IAmRequest):
            # self._log.debug("Received IAmRequest from device with ID: %i and address %s:%i",
            #           apdu.iAmDeviceIdentifier[1],
            #           apdu.pduSource.addrTuple[0],
            #           apdu.pduSource.addrTuple[1]
            #           )

@samson0v  @imbeacon FYI. Please suggest on how to proceed further.

From my perspective, this works for me. and nothing existing is broken.

samson0v, 2024-12-03:
Hi @knityanand and everyone!
We are pleased to inform you that we added a new Async BACnet connector that used bacpypes3 (via PR #1607).
The old BACnet connector is now deprecated and can only be enabled manually.
Please, run the latest version of the gateway via the master branch and let us know about the results.

imbeacon, 2024-12-11:
This issue closed, because relates to outdated implementation of connector, please try the latest version (3.6.2), and feel free to open a new issue if you discover some misbehaviour.

closed 2024-12-11 by imbeacon