thingsboard/thingsboard-gateway #1482: [BUG]

ashdam, 2024-08-05:
**Describe the bug**
We got the following message from logs 

[image]

with the following configuration (Some sensitive fields have been anonymized)

[image]

```{
  "server": {
    "name": "MB107_UDC1",
    "url": "opc.tcp://xxxxxxxxxxx:59300/xxxxxx/UnifiedDataCollector",
    "timeoutInMillis": 5000,
    "scanPeriodInMillis": 5000,
    "disableSubscriptions": true,
    "subCheckPeriodInMillis": 100,
    "showMap": false,
    "security": "Basic128Rsa15",
    "identity": {
      "type": "anonymous"
    },
    "mapping": [
      {
        "deviceNodePattern": "Root\\.Objects\\.Server",
        "deviceNamePattern": "MB107_UDC1",
        "attributes": [
          {
            "key": "serverUai",
            "path": "${ns=2;b=sbnFUIh4jWVrbbrxt463og==}"
          },
          {
            "key": "deviceUai",
            "path": "${ns=2;b=sbnFUIh4jWVrbbrxt463og==}"
          },
          {
            "key": "siteCode",
            "path": "${ns=2;b=zCGYLEQhEXoiQHQBVhcIoA==}"
          },
          {
            "key": "line",
            "path": "${ns=2;b=WrO0W3muBdo+gdCYlWooSQ==}"
          }
        ],
        "timeseries": [
          {
            "key": "/FileSystem/FileSystemFileCountLimit",
            "path": "${ns=80;s=FileSystemFileCountLimit}"
          },
          {
            "key": "/FileSystem/FileSystemFileCount",
            "path": "${ns=80;s=FileSystemFileCount}"
          },
          **{
            "key": "/Server/ServerDiagnostics/Udc/License/Status",
            "path": "${ns=3;i=4}"
          },**
          {
            "key": "/Server/ServerStatus",
            "path": "${ns=0;i=2256}"
          }
        ]
      }
    ]
  },
  "logLevel": "ERROR",
  "name": "MB107_UDC1_connector",
  "enableRemoteLogging": true,
  "id": "727e540a-4526-4644-9918-77501a379e14"
}
```
The key that breaks everything up is /Server/ServerDiagnostics/Udc/License/Status which contains the following information extracted from UAExpert.

**Connector name (If bug in the some connector):**
OPC-UA Asyncua Connector

**Error traceback (If available):**
```
2024-08-05 16:28:56 - |ERROR| - [tb_gateway_service.py] - tb_gateway_service - __send_to_storage - 1103 - Object of type StatusCode is not JSON serializable
2024-08-05 16:29:01 - |ERROR| - [tb_gateway_service.py] - tb_gateway_service - __send_to_storage - 1103 - Object of type StatusCode is not JSON serializable
2024-08-05 16:29:06 - |ERROR| - [tb_gateway_service.py] - tb_gateway_service - __send_to_storage - 1103 - Object of type StatusCode is not JSON serializable
2024-08-05 16:29:11 - |ERROR| - [tb_gateway_service.py] - tb_gateway_service - __send_to_storage - 1103 - Object of type StatusCode is not JSON serializable
```

**Versions (please complete the following information):**
 - OS:  Ubuntu 22.04
 - Thingsboard IoT Gateway version 3.5.1
 - Python version: 3.7

imbeacon, 2024-08-06:
Hi @ashdam,

Thank you for your interest in ThingsBoard IoT Gateway. We was able to reproduce the issue and add a fix to master branch - https://github.com/thingsboard/thingsboard-gateway/commit/2030c90ae94c3103019ccb62e3b5199fb499e936

You can try to use a master branch or just add this commit via cherry-pick to release version. Or wait for new gateway release.

ashdam, 2024-08-06:
It worked! 
[image]

Thank you!

closed 2024-08-06 by ashdam