thingsboard/thingsboard-gateway #1484: [HELP] IoTGW Performance

ashdam, 2024-08-06:
**Describe the issue**
We are trying to stress the platform. In order to do this, we started with an OPC-UA async connector with 1000 timeseries declared and "polling" (not subscribe) every second.
Information is sent this way: IoTGateway ==> Edge PE ==> TB PE
We are getting the following information from statistics:
[image]

110 events /s, memory size increasing 50 MB, but networkfraffic hasn't reached 1MB/s
[image]

Do you know what mabe be happening?

**Configuration** (Attach your configuration file)
```
version: '3.4'
services:
  # ThingsBoard IoT Gateway Service Configuration
  tb-gateway:
    image: thingsboard/tb-gateway
    container_name: tb-gateway
    restart: always

    # Ports bindings - required by some connectors
    ports:
        - "5000:5000" # Comment if you don't use REST connector and change if you use another port
        # Uncomment and modify the following ports based on connector usage:
#        - "1052:1052" # BACnet connector
#        - "5026:5026" # Modbus TCP connector (Modbus Slave)
#        - "50000:50000/tcp" # Socket connector with type TCP
#        - "50000:50000/udp" # Socket connector with type UDP

    # Necessary mapping for Linux
    extra_hosts:
      - "host.docker.internal:host-gateway"

    # Environment variables
    environment:
      - host=xx.xxx.xxx.xx
      - port=1883
      - accessToken=xxxxxxxxxxxxxxxxxxx

    # Volumes bind
    volumes:
      - tb-gw-config:/thingsboard_gateway/config
      - tb-gw-logs:/thingsboard_gateway/logs
      - tb-gw-extensions:/thingsboard_gateway/extensions

# Volumes declaration for configurations, extensions and configuration
volumes:
  tb-gw-config:
    name: tb-gw-config
  tb-gw-logs:
    name: tb-gw-logs
  tb-gw-extensions:
    name: tb-gw-extensions

```

**Connector name (If you need help with some connector/converter):**
[e.g. MQTT Connector]

**Error traceback (If it was raised):**
```
'deviceName'
Traceback (most recent call last):
  File "<input>", line 2, in <module>
KeyError: 'deviceName'
```

**Versions (please complete the following information):**
 - OS: Ubuntu 22
 - Thingsboard IoT Gateway version 3.5.1
 - Python version: 3.12

ashdam, 2024-08-06:
@imbeacon Hi, may you explain this two parameters you have in documentation ? whats the difference between the two
[image]

https://thingsboard.io/docs/iot-gateway/configuration/

ashdam, 2024-08-06:
Alex, help me with this and we have made sure TB PE has not rate limits "Not set"
[image]

Then configured GW with the following parameters with no success.

[image]

ashdam, 2024-08-14:
We made a test where edge is not involved so gateway is directly connected to TB in the cloud.
Here is the i/o usage 
[image]

We changed storage from memory to filesystem to check if there were data accumulation and there wasn't
[image]

The performance got a lil bit better 
[image]

Our current conclusion is: OPC-UA asyncio connector is not enough efficient /performant

samson0v, 2024-08-14:
@ashdam if you increase maxPayloadSizeBytes, one MQTT message can have multiple device messages, so measuring performance with this metric is not effective.
Does all data from devices come to ThingsBoard or do you have data loss?

ashdam, 2024-08-14:
We already did 
[image]

Yes, only thingsboard gateway is in this machine and go to TB cloud

ashdam, 2024-08-17:
This has been solved along with many other upgrades, improvements and stabilizations in https://github.com/thingsboard/thingsboard-gateway/pull/1501

closed 2024-08-17 by ashdam