thingsboard/thingsboard-gateway #1485: [BUG] exception - 127 - Node status code error: The requested operation has no match to return.(BadNoMatch) and other

ashdam, 2024-08-06:
**Describe the bug**

First of all, thank you for your support, guys ;) we have doing the stress tests right now

We have detected in our devices that some OPC-UA nodes some times are "offline" or not available temporally. In our current system this is something that may happens and it recovers when someone fixes the node.

We have observed the following errors after loading a big connector file with hundred of measures in an iotgateway:

[image]
[image]
[image]
[image]

We have detected the following behaviour following this errors:

- No other (the rest of nodes and variables) data is sent while these nodes are configured. Would be nice the rest of telemetry is sent.
- DEBUG: Log doesn't blame the exact node with this problem so we have to figure it between hundreds.
- Any modification (in the TB UI through "Update" button) after this point is useless since the GW becomes unreachable as throwing the following error:
[image]

**configuration yml**:

[image]

** full connector file **
[full_connector.json](https://github.com/user-attachments/files/16511865/full_connector.json)

**Connector name (If bug in the some connector):**
[OPC-UA asyncua Connector]

**Error traceback (If available):**
in the upper screens

**Versions (please complete the following information):**
 - OS: Ubuntu 22
 - Thingsboard IoT Gateway version: 3.5.1 (with this commit based on master https://github.com/thingsboard/thingsboard-gateway/commit/2030c90ae94c3103019ccb62e3b5199fb499e936 )
 - Python version: 3.12

closed 2024-08-07 by imbeacon

ashdam, 2024-08-09:
@imbeacon 
Looking at the code , i noticed that you modified **thingsboard_gateway/connectors/opcua/opcua_connector.py** file but the issue was reported on opcua async. shouldn't this be taken to thingsboard_gateway/connectors/**opcua_asyncio**/opcua_connector.py?

samson0v, 2024-08-13:
@ashdam, with the last release, we abandoned synchronous opcua, so all opcua async code was moved to opcua.

ashdam, 2024-08-14:
Understood ;)