thingsboard/thingsboard-gateway #1486: [BUG] Deletion of a running device from TB Edge UI produces endless device creation

ashdam, 2024-08-06:
**Describe the bug**
Delete a working device from TB Edge UI produces endless device creation.

Steps to reproduce:

1.  Start a connector (we used opc-ua async one)
2. Go to http://localhost:8080/entities/devices/all 
3. Delete the device while its running 
4. It starts creating endless devices 

[image]

[image]

**Connector name (If bug in the some connector):**
OPC-UA Asyncua Connector

**Error traceback (If available):**
shown above

**Versions (please complete the following information):**
 - OS: Ubuntu 22
 - Thingsboard IoT Gateway version 3.5.1
 - Python version: 3.12

samson0v, 2024-08-07:
Hi @ashdam, thanks for your interest in ThingsBoard IoT Gateway!
This is a normal and predictable situation, because the data from this device has not yet been sent, and it is not correct to delete this data from the storage.
So the following situation arises when the device is deleted, but the data from it is reading from the storage and the device must be created again.

ashdam, 2024-08-07:
Hello @samson0v  

I did not explain myself very well :( sorry.

I agree not to lose data but, the behaviour of creating devices endlessly over and over with no human interaction, is neither desirable imho. I can live-demo the behaviour in case you need it :) last time i tried it created hundreds of devices alone till i stopped it.
I can understand that the remaining data been sent create a new device but only one, not hundreds.

On the other hand if this is something user shouldn't do it, shouldn't be better to prevent/warn user from deleting something is running, isn't it? 

Regards

ashdam, 2024-08-13:
Tried to reproduce it but couldn't this time. New exception arose.

2024-08-13 08:15:56 - |INFO| - [tb_gateway_service.py] - tb_gateway_service - __rpc_gateway_processing - 1339 - Received RPC request to the gateway, id: 0, method: gateway_device_deleted
2024-08-13 08:15:56 - |INFO| - [tb_gateway_service.py] - tb_gateway_service - __rpc_gateway_processing - 1347 - Remote shell is disabled.
2024-08-13 08:15:56 - |INFO| - [tb_gateway_service.py] - tb_gateway_service - __process_deleted_gateway_devices - 659 - Received deleted gateway device notification: MB107_UDC1_A0000K9J76
2024-08-13 08:15:59 - |ERROR| - [tb_logger.py] - tb_logger - exception - 133 - 'bool' object is not iterable
Traceback (most recent call last):
  File "/home/admin/tb-iot-gateway-latest/thingsboard-gateway/venv/lib/python3.12/site-packages/thingsboard_gateway-3.5.1-py3.12.egg/thingsboard_gateway/gateway/tb_gateway_service.py", line 1438, in __send_rpc_reply
    self.tb_client.client.send_rpc_reply(req_id, content, quality_of_service=quality_of_service,
  File "/home/admin/tb-iot-gateway-latest/thingsboard-gateway/venv/lib/python3.12/site-packages/tb_device_mqtt.py", line 596, in send_rpc_reply
    info = self._publish_data(resp, RPC_RESPONSE_TOPIC + req_id, quality_of_service)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/admin/tb-iot-gateway-latest/thingsboard-gateway/venv/lib/python3.12/site-packages/tb_device_mqtt.py", line 749, in _publish_data
    return self._send_request(TBSendMethod.PUBLISH, {"topic": topic, "payload": data, "qos": qos}, timeout,
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/admin/tb-iot-gateway-latest/thingsboard-gateway/venv/lib/python3.12/site-packages/tb_device_mqtt.py", line 668, in _send_request
    return self.__send_publish_with_limitations(kwargs, timeout, device, msg_rate_limit, dp_rate_limit)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/admin/tb-iot-gateway-latest/thingsboard-gateway/venv/lib/python3.12/site-packages/tb_device_mqtt.py", line 682, in __send_publish_with_limitations
    datapoints = self._count_datapoints_in_message(data_for_analysis, device=device)
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/admin/tb-iot-gateway-latest/thingsboard-gateway/venv/lib/python3.12/site-packages/tb_device_mqtt.py", line 875, in _count_datapoints_in_message
    for item in data:
TypeError: 'bool' object is not iterable

closed 2024-08-21 by imbeacon