thingsboard/thingsboard-gateway #1488: Trying to use api to send the RPC command and getting the server_side_rpc_handler error.[HELP]

Vaghulkumar95, 2024-08-07:
**Describe the issue**
Create description about your issue, and your actions to solve it.
We are trying to change the value Feedback_Source_Select via the api https://thingsboard.cloud:443/api/plugins/rpc/twoway/5de2aaa0-4d90-11ef-847b-9766e2a793fa
[image]
Sometimes we are getting the 408 Timeout request
[image]
Sometimes we are getting the 409 Conflict
After restarting thingsboard it is okay.
[image]
Can you please help us understand why we are unable to send RPC requests successfully?

**Configuration** (Attach your configuration file)
**Notate: Remove Access token from file if you want to attach a tb_gateway.yaml**
[config.txt](https://github.com/user-attachments/files/16512397/config.txt)

**Connector name (If you need help with some connector/converter):**
[e.g. MQTT Connector]

**Error traceback (If it was raised):**
```
Traceback (most recent call last):
File "/usr/local/lib/python3.10/site-packages/thingsboard_gateway/connectors/modbus/modbus_connector.py", line 652, in server_side_rpc_handler
device = ModbusConnector.__get_device_by_name(server_rpc_request[DEVICE_SECTION_PARAMETER],
File "/usr/local/lib/python3.10/site-packages/thingsboard_gateway/connectors/modbus/modbus_connector.py", line 763, in __get_device_by_name
return tuple(filter(lambda slave: slave.device_name == device_name, devices))[0]
```

**Versions (please complete the following information):**
 - OS: [Debian 11]
 - Thingsboard IoT Gateway version [3.5.1]
 - Python version [Python 3.10.12]

Vaghulkumar95, 2024-08-09:
Hi @samson0v @imbeacon , For the same hub I'm getting the below errors now.
File "/usr/local/lib/python3.10/site-packages/thingsboard_gateway/connectors/modbus/modbus_connector.py", line 590, in __fun>
value=config[PAYLOAD_PARAMETER],
KeyError: 'payload'
10:02:45 ERROR tb_logger:127 'payload'
Traceback (most recent call last):
File "/usr/local/lib/python3.10/site-packages/thingsboard_gateway/connectors/modbus/modbus_connector.py", line 405, in __pro>
input_data = self.__function_to_device(device, current_data)
File "/usr/local/lib/python3.10/site-packages/thingsboard_gateway/connectors/modbus/modbus_connector.py", line 590, in __fun>
value=config[PAYLOAD_PARAMETER],
KeyError: 'payload'

Started ThingsBoard Gateway.
14:54:14 ERROR tb_logger:127 'config'
Traceback (most recent call last):
File "/usr/local/lib/python3.10/site-packages/thingsboard_gateway/gateway/tb_gateway_service.py", line 589, in __init_remote>
self.__remote_configurator = RemoteConfigurator(self, self.__config)
File "/usr/local/lib/python3.10/site-packages/thingsboard_gateway/tb_utility/tb_gateway_remote_configurator.py", line 42, in>
self._load_connectors_configuration()
File "/usr/local/lib/python3.10/site-packages/thingsboard_gateway/tb_utility/tb_gateway_remote_configurator.py", line 180, i>
config = connector.pop('config')[general_connector_config['configuration']]
KeyError: 'config'

the hub was collecting the data fine and suddenly we got these errors. Can you also explain what is wrong here?

Vaghulkumar95, 2024-08-20:
Hi @samson0v @imbeacon, Do we have any updates on this?

Vaghulkumar95, 2024-08-22:
Hi @samson0v @imbeacon ,

We are getting the below error, Can you please check on this ?

File "/usr/local/lib/python3.10/site-packages/thingsboard_gateway/connectors/modbus/bytes_modbus_uplink_converter.py", line>
decoded_data = self.decode_from_registers(decoder, configuration)
File "/usr/local/lib/python3.10/site-packages/thingsboard_gateway/connectors/modbus/bytes_modbus_uplink_converter.py", line>
decoded = decoder_functions[lower_type]()
File "/root/.local/lib/python3.10/site-packages/pymodbus/payload.py", line 403, in decode_32bit_uint
handle = self._unpack_words(fstring, handle)
File "/root/.local/lib/python3.10/site-packages/pymodbus/payload.py", line 359, in _unpack_words
handle = unpack(f"!{wc_value}H", handle)
struct.error: unpack requires a buffer of 4 bytes

imbeacon, 2025-07-08:
Hi @Vaghulkumar95,

RPC to Modbus connector was fixed in the latest release, please try to use it. Feel free to open a new issue in case of discovering any misbehaving.

closed 2025-07-08 by imbeacon