thingsboard/thingsboard-gateway #1505: [BUG]

ultimate-smile, 2024-08-16:
**Describe the bug**
[image]
The above error message appeared after the gateway ran for a while; the  connector  used opcua.
**Connector name (If bug in the some connector):**
opcua connector

**Error traceback (If available):**
```
'deviceName'
2024-08-14 08:47:38 - |INFO| - [tb_updater.py] - tb_updater - check_for_new_version - 85 - 3.5.1
2024-08-14 08:47:38 - |INFO| - [tb_updater.py] - tb_updater - check_for_new_version - 87 -

[===UPDATE===]

 New version 3.5.1 is available!

[===UPDATE===]

2024-08-14 09:17:02 - |ERROR| - [client.py] - client - _monitor_server_loop - 559 - Error in watchdog loop
Traceback (most recent call last):
  File "/usr/local/lib/python3.11/asyncio/tasks.py", line 500, in wait_for
    return fut.result()
           ^^^^^^^^^^^^
asyncio.exceptions.CancelledError

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/root/.local/lib/python3.11/site-packages/asyncua/client/client.py", line 552, in _monitor_server_loop
    _ = await self.nodes.server_state.read_value()
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/.local/lib/python3.11/site-packages/asyncua/common/node.py", line 207, in read_value
    result = await self.read_data_value()
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/.local/lib/python3.11/site-packages/asyncua/common/node.py", line 220, in read_data_value
    return await self.read_attribute(ua.AttributeIds.Value, None, raise_on_bad_status)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/.local/lib/python3.11/site-packages/asyncua/common/node.py", line 342, in read_attribute
    result = await self.session.read(params)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/.local/lib/python3.11/site-packages/asyncua/client/ua_client.py", line 404, in read
    data = await self.protocol.send_request(request)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/.local/lib/python3.11/site-packages/asyncua/client/ua_client.py", line 167, in send_request
    data = await asyncio.wait_for(self._send_request(request, timeout, message_type), timeout if timeout else None)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/asyncio/tasks.py", line 502, in wait_for
    raise exceptions.TimeoutError() from exc
TimeoutError
2024-08-14 09:17:04 - |WARNING| - [opcua_connector.py] - opcua_connector - start_client - 189 - Failed to connect opcua
2024-08-14 09:17:06 - |WARNING| - [opcua_connector.py] - opcua_connector - start_client - 189 - Failed to connect opcua
2024-08-14 09:17:08 - |WARNING| - [opcua_connector.py] - opcua_connector - start_client - 189 - Failed to connect opcua
2024-08-14 09:17:10 - |WARNING| - [opcua_connector.py] - opcua_connector - start_client - 189 - Failed to connect opcua
2024-08-14 09:17:12 - |WARNING| - [opcua_connector.py] - opcua_connector - start_client - 189 - Failed to connect opcua
2024-08-14 09:47:39 - |INFO| - [tb_updater.py] - tb_updater - check_for_new_version - 85 - 3.5.1
2024-08-14 09:47:39 - |INFO| - [tb_updater.py] - tb_updater - check_for_new_version - 87 -

[===UPDATE===]

 New version 3.5.1 is available!

[===UPDATE===]

2024-08-14 10:47:41 - |INFO| - [tb_updater.py] - tb_updater - check_for_new_version - 85 - 3.5.1
2024-08-14 10:47:41 - |INFO| - [tb_updater.py] - tb_updater - check_for_new_version - 87 -
KeyError: 'deviceName'
```

**Versions (please complete the following information):**
 - OS: [e.g. Ubuntu 18.04]
 - Thingsboard IoT Gateway version [e.g. 2.0]
 - Python version[e.g. 3.7]

samson0v, 2024-08-16:
Hi @ultimate-smile, thanks for your interest in ThingsBoard IoT Gateway!
We have a big performance and general update for the OPC-UA connector, so this [PR](https://github.com/thingsboard/thingsboard-gateway/pull/1501) can resolve your problem.

ashdam, 2024-08-17:
The error seems to be just a timeout on the main loop reading from OPCUA server. Networking issue ?

closed 2024-08-21 by imbeacon