thingsboard/thingsboard-gateway #1513: [HELP] How to connect tb-gateway to external broker (without using thingsboard)

mgruenberger21, 2024-08-28:
Hello,

is it possible to use thingsboard-gateway for publishing the collected data (modbus, bacnet, etc.) to an external mqtt-broker (e.g. mosquitto) without connecting it to a thingsboard server?
In the tb_gateway.json I usually define a thingsboard-server for mqtt-pub. Something like:

{
  "thingsboard": {
    "host": "X.X.X.X",
    "port": 1883,
    "remoteShell": false,
    "remoteConfiguration": true,
    "statistics": {
      "enable": true,
      "statsSendPeriodInSeconds": 60
    },
    "deviceFiltering": {
      "enable": false,
      "filterFile": "list.json"
    },
    "maxPayloadSizeBytes": 1024,
    "minPackSendDelayMS": 60,
    "minPackSizeToSend": 500,
    "checkConnectorsConfigurationInSeconds": 10,
    "handleDeviceRenaming": true,
        "security": {
                "clientId": "",
                "username": "",
                "password": ""

Is it possible to publish this to another (non-thingsboard) broker using mqtt?
And how can I configure this in tb_gateway.json file?

**Versions**
 - OS: Rocky-Linux 9.4
 - Thingsboard IoT Gateway version 3.5.1-1 (rpm-package)
 - Python version 3.9

samson0v, 2024-08-28:
Hi @mgruenberger21, thanks for your interest in ThingsBoard IoT Gateway!
Unfortunately, the gateway can only be used with ThingsBoard.

mgruenberger21, 2024-08-28:
Hi,
thank you for your message. As far as I understood, publishing IoT-gateway-messages is just possible to the thingsboard-ce aund pe?
Not the thingsboard-mqtt-broker (tbmq)?

samson0v, 2024-08-28:
@mgruenberger21 yes

mgruenberger21, 2024-08-28:
OK thank you.
Do you see any other solution to send this data to another broker?
Maybe installing thingsboard CE on the same machine, collecting the data there and publish them via mqtt to mosquitto or tbmq-broker?
Or subscribing the data of thingsboard-ce on the broker?

ellenfel, 2024-08-29:
@mgruenberger21 

I had a similar problem in a project. I think best solution to this is to like you said also Thingsboard CE and then use MQTT API to publish collected messages. 

You can check thingsboard's api here: [swagger](https://demo.thingsboard.io/swagger-ui/index.html)

here's the script we used at the time with some device id's changed.

```
#!/bin/bash

# MQTT Configuration
MQTT_HOST="127.0.0.1"
MQTT_PORT="1883"

# Define MQTT Topics
MQTT_TOPICS=(
    "topic1/status"
    "topic2/status"
    "topic3/status"
    "topic4/status"
    "topic5/status"
)

# Define ThingsBoard URLs
THINGSBOARD_URLS=(
    "http://192.168.1.44:8080/api/v1/1413414/telemetry"
    "http://192.168.1.44:8080/api/v1/1234123412/telemetry"
    "http://192.168.1.44:8080/api/v1/123412341234/telemetry"
    "http://192.168.1.44:8080/api/v1/13412341234/telemetry"
    "http://192.168.1.44:8080/api/v1/13412341234/telemetry"
)

# Subscribe to MQTT topics
for ((i=0; i<${#MQTT_TOPICS[@]}; i++)); do
    mosquitto_sub -h $MQTT_HOST -p $MQTT_PORT -t ${MQTT_TOPICS[$i]} | while read -r message; do
        # Execute curl command with the received message and corresponding ThingsBoard URL
        curl -v -X POST ${THINGSBOARD_URLS[$i]} --header "Content-Type: application/json" --data "{\"message\": \"$message\"}"
    done &
done

# Wait for all background processes to finish
wait

```

similarly you need to set up a script as a service for publishing readied telemetry from your devices. or make a rulechain that does that.

we were also needing dashboard for monitoring and thats why we  go with Thingsboard tho. if you are only going to use communication console i would recommend Node-Red instead.

ZhaoLinl, 2024-09-10:
> 你好@mgruenberger21，感谢您对 ThingsBoard IoT 网关的关注！ 遗憾的是，该网关只能与 ThingsBoard 一起使用。

Can I get gateway data from Thingsboard through the API?

closed 2025-04-16 by imbeacon