thingsboard/thingsboard-gateway #1514: [HELP]Modbus_serial cannot configure multiple devices.

ZhaoLinl, 2024-08-30:
**Describe the issue**
Create description about your issue, and your actions to solve it.
Modbus_serial cannot configure multiple devices.Currently, only one device ID can be configured.How to solve the problem that we need to connect multiple devices on one interface?
Looking forward to your reply.

**Configuration** (Attach your configuration file)
**Notate: Remove Access token from file if you want to attach a tb_gateway.yaml**

**Connector name (If you need help with some connector/converter):**
[e.g. MQTT Connector]

**Error traceback (If it was raised):**
```
'deviceName'
Traceback (most recent call last):
  File "<input>", line 2, in <module>
KeyError: 'deviceName'
```

**Versions (please complete the following information):**
 - OS: [e.g. Ubuntu 20.04]
 - Thingsboard IoT Gateway version [e.g. 3.4.6]
 - Python version[e.g. 3.9]

ZhaoLinl, 2024-08-30:
[image]
[image]

KashifAhmd, 2024-09-02:
I don't want to do it manually, Is there any API endpoint to configure new device?