thingsboard/thingsboard-gateway #1515: [HELP]

itwish, 2024-09-02:
**Describe the issue**
Hello, I follow this doc:
https://thingsboard.io/docs/iot-gateway/getting-started/
but the mqtt connector can't communicate with mqtt broker (emqx), network is ok , firewall is closed. Thingsboard is running in  ide (idea).
My Gateway status is  Active. Mqtt connector(ems-mqtt) logs is empty.

**Configuration** 
tb_gateway.json:
`
{
  "thingsboard": {
    "host": "192.168.18.18",
    "port": 1883,
    "remoteShell": false,
    "remoteConfiguration": true,
    "security": {
      "type": "accessToken",
      "accessToken": "***"
    },
    "statistics": {
      "enable": true,
      "statsSendPeriodInSeconds": 3600
    },
    "deviceFiltering": {
      "enable": false,
      "filterFile": "list.json"
    },
    "maxPayloadSizeBytes": 1024,
    "minPackSendDelayMS": 200,
    "minPackSizeToSend": 500,
    "checkConnectorsConfigurationInSeconds": 60,
    "handleDeviceRenaming": true,
    "qos": 1,
    "checkingDeviceActivity": {
      "checkDeviceInactivity": false,
      "inactivityTimeoutSeconds": 200,
      "inactivityCheckPeriodSeconds": 500
    },
    "ts": 1725001579840,
    "rateLimits": "DEFAULT_RATE_LIMIT",
    "dpRateLimits": "DEFAULT_RATE_LIMIT"
  },
  "storage": {
    "type": "memory",
    "read_records_count": 100,
    "max_records_count": 100000,
    "data_folder_path": "./data/",
    "max_file_count": 10,
    "max_read_records_count": 10,
    "max_records_per_file": 10000,
    "data_file_path": "./data/data.db",
    "messages_ttl_check_in_hours": 1,
    "messages_ttl_in_days": 7,
    "ts": 1725001579840
  },
"grpc": {
    "enabled": false,
    "serverPort": 9595,
    "keepaliveTimeMs": 10000,
    "keepaliveTimeoutMs": 5000,
    "keepalivePermitWithoutCalls": true,
    "maxPingsWithoutData": 0,
    "minTimeBetweenPingsMs": 10000,
    "minPingIntervalWithoutDataMs": 5000,
    "keepAliveTimeMs": 10000,
    "keepAliveTimeoutMs": 5000
  },
  "connectors": [
    {
      "type": "mqtt",
      "name": "ems-mqtt",
      "configuration": "emsMqtt.json"
    }
  ]
}                                                                                                                                                                                                                   
`

emsMqtt.json:
`{
  "broker": {
    "name": "",
    "host": "192.168.18.52",
    "port": 1883,
    "version": 5,
    "clientId": "tb_gateway1",
    "maxNumberOfWorkers": 10,
    "maxMessageNumberPerWorker": 100
  },
  "logLevel": "INFO",
  "name": "ems-mqtt",
  "enableRemoteLogging": false,
  "id": "e37943d8-36cd-4f02-9931-ee3c88d680c3",
  "dataMapping": [
    {
      "topicFilter": "data/",
      "subscriptionQos": 0,
      "converter": {
        "type": "json",
        "deviceInfo": {
          "deviceNameExpression": "Device Demo",
          "deviceNameExpressionSource": "constant",
          "deviceProfileExpressionSource": "constant",
          "deviceProfileExpression": "default"
        },
        "attributes": [
          {
            "key": "frequency",
            "value": "${frequency}",
            "type": "integer"
          },
          {
            "key": "power",
            "value": "${power}",
            "type": "integer"
          }
        ],
        "timeseries": [
          {
            "key": "temperature",
            "value": "${temperature}",
            "type": "integer"
          },
         {
            "key": "humidity",
            "value": "${humidity}",
            "type": "integer"
          }
        ]
      }
    }
  ],
  "requestsMapping": [
    {
      "requestType": "connectRequests",
      "requestValue": {
        "topicFilter": "sensor/connect",
        "deviceInfo": {
          "deviceNameExpression": "${SerialNumber}",
          "deviceNameExpressionSource": "message",
          "deviceProfileExpressionSource": "constant",
          "deviceProfileExpression": "default"
        }
      }
    },
    {
      "requestType": "connectRequests",
      "requestValue": {
        "topicFilter": "sensor/+/connect",
        "deviceInfo": {
          "deviceNameExpression": "(?<=sensor\\/)(.*?)(?=\\/connect)",
          "deviceNameExpressionSource": "topic",
          "deviceProfileExpressionSource": "constant",
          "deviceProfileExpression": "default"
        }
      }
    },
    {
      "requestType": "disconnectRequests",
      "requestValue": {
        "topicFilter": "sensor/disconnect",
        "deviceInfo": {
          "deviceNameExpression": "${SerialNumber}",
          "deviceNameExpressionSource": "message"
        }
      }
    },
{
      "requestType": "disconnectRequests",
      "requestValue": {
        "topicFilter": "sensor/+/disconnect",
        "deviceInfo": {
          "deviceNameExpression": "(?<=sensor\\/)(.*?)(?=\\/disconnect)",
          "deviceNameExpressionSource": "topic"
        }
      }
    }
  ],
  "workers": {
    "maxNumberOfWorkers": 10,
    "maxMessageNumberPerWorker": 100
  }
`

**Connector name (If you need help with some connector/converter):**
MQTT Connector

**Error traceback:**
```
tb-gateway | 2024-09-02 01:17:36 - |ERROR| - [tb_logger.py] - tb_logger - exception - 127 - 'list' object has no attribute 'get'
tb-gateway | Traceback (most recent call last):
tb-gateway | File "/thingsboard_gateway/gateway/tb_gateway_service.py", line 912, in __connect_with_connectors
tb-gateway | connector = self._implemented_connectors[connector_type](self,
tb-gateway | ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
tb-gateway | File "/thingsboard_gateway/connectors/mqtt/mqtt_connector.py", line 162, in __init__
tb-gateway | self.load_handlers('serverSideRpc', mandatory_keys['serverSideRpc'], self.__server_side_rpc)
tb-gateway | File "/thingsboard_gateway/connectors/mqtt/mqtt_connector.py", line 260, in load_handlers
tb-gateway | config = self.config.get(handler_flavor) or self.config.get("requestsMapping", {}).get(handler_flavor)
tb-gateway | ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
tb-gateway | AttributeError: 'list' object has no attribute 'get'
```

**Versions :**
 - OS: Debian GNU/Linux 12
 - Thingsboard IoT Gateway version: 3.5.1
 - Python version: 3.11.9

imbeacon, 2024-09-02:
Hi @itwish,

Thank you for your interest in ThingsBoard IoT Gateway, looks like you are using a new configuration format, but old gateway, probably not the latest version from the master. Please try to update the code to version from master branch and try again. Let us know if you still have issues.

itwish, 2024-09-02:
Thanks for your reply. I have confirmed that I am using the latest gateway version: 3.5.1. I installed the latest gateway by docker. Do you mean it must be installed with latest source code?

itwish, 2024-09-02:
Hi @imbeacon,
Thanks for your reply. I have confirmed that I am using the latest gateway version: 3.5.1. I installed the latest gateway by docker. Do you mean it must be installed with latest source code?

imbeacon, 2024-09-02:
There are some modifications in ThingsBoard gateway dashboard in master branch, the gateway in master branch is adopted to them, but the old gateway cannot handle them. so in order to use version of ThingsBoard form the master branch - you also need to use the newest gateway. We are in progress to add configuration conversion of the gateway configuration on the gateway dashboard, but it is not added yet to ThingsBoard.
You can also try to configure the connector from basic configuration tab and probably you will see the difference between configs (serverSideRpc, attributeUpdates, etc - won’t be in array, they will be in object requestMapping).

itwish, 2024-09-02:
Great! The problem has been solved，thanks for your help.

wish the world be like open source world

closed 2024-09-03 by imbeacon