thingsboard/thingsboard-gateway #1517: [HELP]I don't want to do configuration manually, Is there any API endpoint to configure new device?

KashifAhmd, 2024-09-03:


imbeacon, 2024-09-03:
Hi @KashifAhmd,

Please provide more information about your request, it is not completely clear, what you need.

KashifAhmd, 2024-09-03:
> Hi @KashifAhmd,
> 
> Please provide more information about your request, it is not completely clear, what you need.

Hi @imbeacon,
Thanks for your response

My doubt is If we have to add a new device with the Modbus gateway we need to add the credentials and parameters of that device in the gateway configuration settings, right?

We do that manually and directly in those configuration settings.

So my question was, is there any API request or any other way to add new device details in that configuration setting without going into that configuration settings directly?