thingsboard/thingsboard-gateway #1520: [HELP] Can't access data with Bacnet

mgruenberger21, 2024-09-04:
Hello, 

I cant`t access data of a heating system with bacnet using thingsboard-gateway.
When I use Yabe, I can accesss all the things I want to know.
But when requesting data with thingsboard-gateway, I don`t get any answer of the heating system to my request.
[image]

e.g. the request for this value shold look like this I think:
{
  "general": {
    "objectName": "tb-gateway",
    "address": "0.0.0.0:47808",
    "objectIdentifier": 599,
    "maxApduLengthAccepted": 1476,
    "segmentationSupported": "segmentedBoth",
    "vendorIdentifier": 15
  },
  "devices": [
    {
      "deviceName": "BACnet Device ${objectName}",
      "deviceType": "default",
      "address": "IPADDR:47808",
      "pollPeriod": 10000,
      "attributes": [
        {
          "key": "Außentemperatur Kessel 1",
          "type": "double",
          "objectId": "analogInput:239",
          "propertyId": "presentValue"
        }
      ]
    }
  ],
  "logLevel": "DEBUG",
  "name": "Bacnet",
  "enableRemoteLogging": true,
  "id": "ac5e542d-623b-432a-9a96-04fe82655e34"
}

Debug-Logs of thingsboard-gateway:

b-gateway  | 2024-09-04 11:50:37 - |INFO| - [tb_gateway_remote_configurator.py] - tb_gateway_remote_configurator - _process_config_request - 204 - Configuration update request received.
tb-gateway  | 2024-09-04 11:50:37 - |DEBUG| - [tb_gateway_remote_configurator.py] - tb_gateway_remote_configurator - _process_config_request - 205 - Got config update request: {'active_connectors': ['Bacnet']}
tb-gateway  | 2024-09-04 11:50:37 - |DEBUG| - [tb_gateway_remote_configurator.py] - tb_gateway_remote_configurator - _handle_active_connectors_update - 404 - Processing active connectors configuration update...
tb-gateway  | 2024-09-04 11:50:37 - |INFO| - [tb_gateway_remote_configurator.py] - tb_gateway_remote_configurator - _process_config_request - 204 - Configuration update request received.
tb-gateway  | 2024-09-04 11:50:37 - |DEBUG| - [tb_gateway_remote_configurator.py] - tb_gateway_remote_configurator - _process_config_request - 205 - Got config update request: {'Bacnet': {'mode': 'basic', 'name': 'Bacnet', 'type': 'bacnet', 'enableRemoteLogging': True, 'logLevel': 'DEBUG', 'sendDataOnlyOnChange': False, 'configuration': 'bacnet.json', 'configurationJson': {'general': {'objectName': 'tb-gateway', 'address': '0.0.0.0:47808', 'objectIdentifier': 599, 'maxApduLengthAccepted': 1476, 'segmentationSupported': 'segmentedBoth', 'vendorIdentifier': 15}, 'devices': [{'deviceName': 'BACnet Device ${objectName}', 'deviceType': 'default', 'address': 'IP-ADDR:47808', 'pollPeriod': 10000, 'attributes': [{'key': 'Außentemperatur Kessel 1', 'type': 'double', 'objectId': 'analogInput:239', 'propertyId': 'presentValue'}]}], 'logLevel': 'DEBUG', 'name': 'Bacnet', 'enableRemoteLogging': True, 'id': 'ac5e542d-623b-432a-9a96-04fe82655e34'}, 'ts': 1725450637146, 'key': 'auto', 'basicConfig': {'general': {'objectName': 'tb-gateway', 'address': '0.0.0.0:47808', 'objectIdentifier': 599, 'maxApduLengthAccepted': 1476, 'segmentationSupported': 'segmentedBoth', 'vendorIdentifier': 15}, 'devices': [{'deviceName': 'BACnet Device ${objectName}', 'deviceType': 'default', 'address': 'IP-ADDR:47808', 'pollPeriod': 10000, 'attributes': [{'key': 'Außentemperatur Kessel 1', 'type': 'double', 'objectId': 'analogInput:239', 'propertyId': 'presentValue'}]}], 'logLevel': 'DEBUG', 'name': 'Bacnet', 'enableRemoteLogging': True, 'id': 'ac5e542d-623b-432a-9a96-04fe82655e34'}}}
tb-gateway  | 2024-09-04 11:50:37 - |DEBUG| - [tb_gateway_remote_configurator.py] - tb_gateway_remote_configurator - _handle_connector_configuration_update - 449 - Processing connectors configuration update...
tb-gateway  | 2024-09-04 11:50:37 - |DEBUG| - [tb_gateway_remote_configurator.py] - tb_gateway_remote_configurator - create_configuration_file_backup - 757 - Backup file created for configuration file bacnet.json in /thingsboard_gateway/config/backup/bacnet.backup.1725450637.json
tb-gateway  | 2024-09-04 11:50:37 - |DEBUG| - [bacnet_connector.py] - bacnet_connector - scan_network - 226 - WhoIsRequest has been sent.
tb-gateway  | 2024-09-04 11:50:37 - |DEBUG| - [bacnet_connector.py] - bacnet_connector - run - 80 - WhoIsRequest has been sent.
tb-gateway  | 2024-09-04 11:50:37 - |DEBUG| - [bacnet_connector.py] - bacnet_connector - scan_network - 226 - WhoIsRequest has been sent.
tb-gateway  | 2024-09-04 11:50:37 - |INFO| - [tb_gateway_remote_configurator.py] - tb_gateway_remote_configurator - _process_config_request - 204 - Configuration update request received.
tb-gateway  | 2024-09-04 11:50:37 - |DEBUG| - [tb_gateway_remote_configurator.py] - tb_gateway_remote_configurator - _process_config_request - 205 - Got config update request: {'Bacnet': {'mode': 'basic', 'name': 'Bacnet', 'type': 'bacnet', 'enableRemoteLogging': True, 'logLevel': 'DEBUG', 'sendDataOnlyOnChange': False, 'configuration': 'bacnet.json', 'configurationJson': {'general': {'objectName': 'tb-gateway', 'address': '0.0.0.0:47808', 'objectIdentifier': 599, 'maxApduLengthAccepted': 1476, 'segmentationSupported': 'segmentedBoth', 'vendorIdentifier': 15}, 'devices': [{'deviceName': 'BACnet Device ${objectName}', 'deviceType': 'default', 'address': 'IP-ADDR:47808', 'pollPeriod': 10000, 'attributes': [{'key': 'Außentemperatur Kessel 1', 'type': 'double', 'objectId': 'analogInput:239', 'propertyId': 'presentValue'}]}], 'logLevel': 'DEBUG', 'name': 'Bacnet', 'enableRemoteLogging': True, 'id': 'ac5e542d-623b-432a-9a96-04fe82655e34'}, 'ts': 1725450637146, 'key': 'auto', 'basicConfig': {'general': {'objectName': 'tb-gateway', 'address': '0.0.0.0:47808', 'objectIdentifier': 599, 'maxApduLengthAccepted': 1476, 'segmentationSupported': 'segmentedBoth', 'vendorIdentifier': 15}, 'devices': [{'deviceName': 'BACnet Device ${objectName}', 'deviceType': 'default', 'address': 'IP-ADDR:47808', 'pollPeriod': 10000, 'attributes': [{'key': 'Außentemperatur Kessel 1', 'type': 'double', 'objectId': 'analogInput:239', 'propertyId': 'presentValue'}]}], 'logLevel': 'DEBUG', 'name': 'Bacnet', 'enableRemoteLogging': True, 'id': 'ac5e542d-623b-432a-9a96-04fe82655e34'}}}

tcpdump:

13:57:32.811805 IP iotgw.bacnet > IPADDR.bacnet: UDP, length 8
13:57:32.814702 IP iotgw.bacnet > IPADDR.bacnet: UDP, length 8

Can someone please help me to solve this problem.?
Sorry, I´m not really a bacnet-specialist.

**Versions (please complete the following information):**
 - OS: Rocky Linux 9.4
 - Thingsboard IoT Gateway version 3.5.1 (Docker)
 - Python version 3.9

HarvUK, 2024-09-08:
Try changing this IP address 

{
"general": {
"objectName": "tb-gateway",
"address": "**0.0.0.0**:47808",

To be the IP address of the network interface of your TBGW.

Harv

mgruenberger21, 2024-09-09:
Hello,

thank you for your message.
Still there is no answer of the Bacnet-Device.
Looks like something is wrong with the request itself. In Wireshark I can see the request sent to bacnet-device, but nothing is sent back:
[image]
Some flags look differnet in yabe-request:
[image]

jurkov, 2024-09-30:
You could try https://github.com/JoelBender/bacpypes/blob/master/samples/ReadObjectList.py to find all the objects. If you don't get your device, your problem might be related to https://github.com/JoelBender/bacpypes.

mgruenberger21, 2024-09-30:
Ok thank you. I just installed bacpypes on my machine and downloaded ReadObjectList.py.
Now the following arguments are required: device_id, device_addr
How, or where can i set this arguments?
device_addr should look like "192.0.0.1:47808" for example?
and what's the  device_id?

mgruenberger21, 2024-10-01:
Thank you jurkov.
With bacpypes I can access all the data. And sorry for my stupid questions, but I`m not a bacnet-specialist :)
But my Bacnet-Device is only responding to "Confirmed-REQ"-messages:
python samples/ReadObjectList.py --ini BACpypes.ini 2XXXXX5 10.XXX.XXXX.1
[image]

When I send "Unconfirmed-REQ who-Is" (as thingsboard-gateway does), I get no answer:
[image]

So I need to make thingsboard to send Confirmed-REQ-messages to the bacnet-device?
How can I do this?
How to specify the device-ID for bacnet-device in thingsboard?

jurkov, 2024-10-04:
You may have to create a feature request or create the feature by yourself. It is possible to create a custom uplink and downlink converter. See https://thingsboard.io/docs/iot-gateway/custom/

I added my BacNET devices based on https://thingsboard.io/docs/iot-gateway/config/bacnet/ line 12 and line 14.

samson0v, 2024-12-03:
Hi @mgruenberger21 and everyone!
We are pleased to inform you that we added a new Async BACnet connector that used bacpypes3 (via PR #1607).
The old BACnet connector is now deprecated and can only be enabled manually.
Please, run the latest version of the gateway via the master branch and let us know about the results.

imbeacon, 2024-12-11:
Hi all,

New implementation of connector was tested with YABE, please try it, for now I close this issue, feel free to open a new one if you discover some misbehaviour with new connector implementation (In gateway 3.6.2+).

closed 2024-12-11 by imbeacon

imbeacon, 2025-05-21:
Hi @ttrading,

This issue was closed, if you discover some issue - please open the new one. Also, please provide your BACnet connector configuration. 

Due to typical issues with BACnet - please check that you set host for application - your IP in the same network, as device and port - 47808, mask usually - 24.

For device - please try to set * instead of host and * instead of port, this will enable global broadcast for device search. Aslo I should mention that you need to run the gateway and emulator on different devices, because emulator may reserver port 47808 and the gateway won't be able to use it for application.