thingsboard/thingsboard-gateway #1530: [HELP]  How to add a new protocol in the Type drop-down box??

DuXiaoChuang, 2024-09-19:
How to add a new protocol in the Type drop-down box??
[image]

yhliyr, 2024-09-19:
perhaps you could try adding a new connector type at [gateway-widget.model.ts-ConnectorType](https://github.com/thingsboard/thingsboard/blob/28986fbf6ec131c925725e2542a0042e14a3122d/ui-ngx/src/app/modules/home/components/widget/lib/gateway/gateway-widget.models.ts#L384) and [gateway-widget.model.ts-GatewayConnectorDefaultTypesTranslatesMap](https://github.com/thingsboard/thingsboard/blob/28986fbf6ec131c925725e2542a0042e14a3122d/ui-ngx/src/app/modules/home/components/widget/lib/gateway/gateway-widget.models.ts#L403) in the thingsboard platform project

DuXiaoChuang, 2024-09-20:
> perhaps you could try adding a new connector type at [...]

OK，thanks，I will try！

DuXiaoChuang, 2024-09-20:
> perhaps you could try adding a new connector type at [...]

How to add a new protocol in the Type drop-down box??
[image]

yhliyr, 2024-09-20:
@DuXiaoChuang You can't do this on the web page, you have to modify the Thingsboard project code as i mentioned above, and compile the ui-ngx project and redeploy the whole project. 在页面上是不能手动添加的，只能改代码重新编译项目打包，改动的位置就是上面我提的两个地方，加两行就行了

or maybe you can try using the CUSTOM type, 或许可以试试 CUSTOM 类型，这个可以指定自定义连接器class的，不过我没试过，我也正在搞自定义connector

DuXiaoChuang, 2024-09-20:
> @DuXiaoChuang You can't do this on the web page, you have to modify the Thingsboard project code as i mentioned above, and compile the ui-ngx project and redeploy the whole project. [...]

我按照你说的加了，没效果啊。

DuXiaoChuang, 2024-09-24:
> @DuXiaoChuang You can't do this on the web page, you have to modify the Thingsboard project code as i mentioned above, and compile the ui-ngx project and redeploy the whole project. [...]

我新增了连接器，命名带上custom，但是这个页签还是没有数据，很奇怪。
[image]
[image]

yhliyr, 2024-09-25:
> > @DuXiaoChuang You can't do this on the web page, you have to modify the Thingsboard project code as i mentioned above, and compile the ui-ngx project and redeploy the whole project. 在页面上是不能手动添加的，只能改代码重新编译项目打包，改动的位置就是上面我提的两个地方，加两行就行了
> > or maybe you can try using the CUSTOM type, 或许可以试试 CUSTOM 类型，这个可以指定自定义连接器class的，不过我没试过，我也正在搞自定义connector
> 
> I added the connector and named it with custom, but this TAB still has no data, which is strange.
> 
> ![image](https://private-user-images.githubusercontent.com/149569575/370116604-1757cf25-87e2-4c48-a6f0-82d8ae9288ec.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjcyNDkzNTgsIm5iZiI6MTcyNzI0OTA1OCwicGF0aCI6Ii8xNDk1Njk1NzUvMzcwMTE2NjA0LTE3NTdjZjI1LTg3ZTItNGM0OC1hNmYwLTgyZDhhZTkyODhlYy5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwOTI1JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDkyNVQwNzI0MThaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT03OWUyYjkzYjY3YTVjMTFjOTA4MDBlMzYxOWEyMDE5YTY3YTJiNTAxM2FhMGVkYjRjNmQ5YWU0Nzg4Y2I0MThhJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.cO7bNnSOg-zQIpNDL3C1Z0s3G9NAYJISmeVxJOQPC2U) ![image](https://private-user-images.githubusercontent.com/149569575/370116667-afc5b603-80a2-47a6-8834-caa73af90aff.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjcyNDkzNTgsIm5iZiI6MTcyNzI0OTA1OCwicGF0aCI6Ii8xNDk1Njk1NzUvMzcwMTE2NjY3LWFmYzViNjAzLTgwYTItNDdhNi04ODM0LWNhYTczYWY5MGFmZi5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwOTI1JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDkyNVQwNzI0MThaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT03MzI5MWI1OGZlNDJjYzQ5YmNhYThjOTAyOTI1ZjkyMmUyZGUxYzI1Yjc2N2VhZDIwYjk0ZTVkMmYxYTQ3OWVlJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.QURBrIjQKtWb3dfEGPUeESsCOzcK0GC5Ke6F0ixA8Ew)

it show "focas", the class name is "focas" right? u can be added ConnectorType in here by "edit component"
这里应该就把类名 focas 当作一种类型了，这里编辑部件可以手动添加连接器类型的

DuXiaoChuang, 2024-09-26:
> > > @DuXiaoChuang You can't do this on the web page, you have to modify the Thingsboard project code as i mentioned above, and compile the ui-ngx project and redeploy the whole project. 在页面上是不能手动添加的，只能改代码重新编译项目打包，改动的位置就是上面我提的两个地方，加两行就行了
> > > or maybe you can try using the CUSTOM type, 或许可以试试 CUSTOM 类型，这个可以指定自定义连接器class的，不过我没试过，我也正在搞自定义connector
> > 
> > 
> > I added the connector and named it with custom, but this TAB still has no data, which is strange.
> > ![image](https://private-user-images.githubusercontent.com/149569575/370116604-1757cf25-87e2-4c48-a6f0-82d8ae9288ec.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjcyNDkzNTgsIm5iZiI6MTcyNzI0OTA1OCwicGF0aCI6Ii8xNDk1Njk1NzUvMzcwMTE2NjA0LTE3NTdjZjI1LTg3ZTItNGM0OC1hNmYwLTgyZDhhZTkyODhlYy5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwOTI1JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDkyNVQwNzI0MThaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT03OWUyYjkzYjY3YTVjMTFjOTA4MDBlMzYxOWEyMDE5YTY3YTJiNTAxM2FhMGVkYjRjNmQ5YWU0Nzg4Y2I0MThhJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.cO7bNnSOg-zQIpNDL3C1Z0s3G9NAYJISmeVxJOQPC2U) ![image](https://private-user-images.githubusercontent.com/149569575/370116667-afc5b603-80a2-47a6-8834-caa73af90aff.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjcyNDkzNTgsIm5iZiI6MTcyNzI0OTA1OCwicGF0aCI6Ii8xNDk1Njk1NzUvMzcwMTE2NjY3LWFmYzViNjAzLTgwYTItNDdhNi04ODM0LWNhYTczYWY5MGFmZi5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwOTI1JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDkyNVQwNzI0MThaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT03MzI5MWI1OGZlNDJjYzQ5YmNhYThjOTAyOTI1ZjkyMmUyZGUxYzI1Yjc2N2VhZDIwYjk0ZTVkMmYxYTQ3OWVlJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.QURBrIjQKtWb3dfEGPUeESsCOzcK0GC5Ke6F0ixA8Ew)
> 
> it show "focas", the class name is "focas" right? You can add a new ConnectorType here by 'Edit Component' 这里应该就把类名 focas 当作一种类型了，这里编辑部件可以手动添加连接器类型的

ok， thanks