thingsboard/thingsboard-gateway #1540: [BUG] docker image thingsboard/tb-gateway is unable to update and throws AttributeError.

jurkov, 2024-09-25:
**Describe the bug**
After the update from 3.5.1 to 3.5.2 the docker container/image of tb-gateway is broken.
It may related to the submodule tb_mqtt_client, as it is a submodule?!

Since today the docker container shows on startup:
```
tb-gateway  | 2024-09-25 06:41:02 - |INFO| - [tb_updater.py] - tb_updater - check_for_new_version - 87 -
tb-gateway  |
tb-gateway  | [===UPDATE===]
tb-gateway  |
tb-gateway  |  New version 3.5.2 is available!
tb-gateway  |
tb-gateway  | [===UPDATE===]
```

**Connector name (If bug in the some connector):**
-

**Error traceback (If available):**
```
tb-gateway  | Traceback (most recent call last):
tb-gateway  |   File "/thingsboard_gateway/gateway/tb_gateway_service.py", line 912, in __connect_with_connectors
tb-gateway  |     connector = self._implemented_connectors[connector_type](self,
tb-gateway  |                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
tb-gateway  |   File "/thingsboard_gateway/connectors/mqtt/mqtt_connector.py", line 133, in __init__
tb-gateway  |     self.__send_data_only_on_change = self.__broker.get(SEND_ON_CHANGE_PARAMETER, DEFAULT_SEND_ON_CHANGE_VALUE)
tb-gateway  | AttributeError: 'NoneType' object has no attribute 'get'
```

**Versions (please complete the following information):**
 - OS: provided by docker image thingsboard/tb-gateway
 - Thingsboard IoT Gateway version 3.5.1
 - Python version provided by docker image thingsboard/tb-gateway

imbeacon, 2024-09-26:
Hi @jurkov,

It doesn’t look related to tb_mqtt_client, the error appeared because “broker” section is missing in mqtt connector configuration. It may be caused due to mismatching in configuration of UI and your existing config. We recommend to use “advanced” tab to configure your connector until ThingsBoard 3.7.1 will be released. It will contain updated UI, that will convert connectors configuration to required format. We are sorry for inconvenience, but unfortunately we cannot provide a new dashboard earlier, because it contains directives that were modified in ThingsBoard 3.7.1.

Also, you can try to create a new MQTT connector and configure it in UI basic mode, it should work properly.
Please, let us know if something wrong.

jurkov, 2024-09-26:
Thank you for your advice. I selected the wrong connector somehow. It's working now.

jurkov, 2024-09-30:
Can be closed.

closed 2024-09-30 by jurkov