thingsboard/thingsboard-gateway #1541: [BUG] Problem with the new version 3.5.2 BAcnet

Isac-enerlab, 2024-09-25:
**the bug**
Testing the new version it stopped reading the variables that we pulled from the operation that runs with version 3.5.1.1.
It reports an error on line 53 of the file (thingsboard_gateway/connectors/bacnet/bacnet_uplink_converter.py", line 53, in convert).
As we have 23 Units with version 3.5.1, we need to fix this bug.

Bacnet Connector

Sep 25 16:41:55  python3[433368]:     fn(self, *args, **kwargs)
Sep 25 16:41:55  python3[433368]:   File "/var/lib/thingsboard_gateway/venv/lib/python3.10/site-packages/thingsboard_gateway/connectors/bacnet/bacnet_utilities/tb_gateway_bacnet_application.py", line 167, in __iam_cb
Sep 25 16:41:55  python3[433368]:     value = self.__connector.default_converters["uplink_converter"]("{}", self._log).convert(None, apdu)
Sep 25 16:41:55  python3[433368]:   File "/var/lib/thingsboard_gateway/venv/lib/python3.10/site-packages/thingsboard_gateway/gateway/statistics/decorators.py", line 31, in inner
Sep 25 16:41:55  python3[433368]:     result = func(*args, **kwargs)
Sep 25 16:41:55  python3[433368]:   File "/var/lib/thingsboard_gateway/venv/lib/python3.10/site-packages/thingsboard_gateway/connectors/bacnet/bacnet_uplink_converter.py", line 53, in convert
Sep 25 16:41:55  python3[433368]:     count=len(dict_result["attributes"]))
Sep 25 16:41:55  python3[433368]: TypeError: string indices must be integers

**Versions**
 - OS: Ubuntu 22.04.4 LTS Memory usage: 11%
 - Thingsboard IoT Gateway version 3.5.2
 - Python version 3.10.12

jurkov, 2024-09-30:
I can confirm the bug. I also got "missing asyncore package". Asyncore is deprecated with Python 3.12. Python 3.12 comes with the latest base docker image. After installing the package by hand (`docker compose exec tb-gateway /bin/bash` and `pip install pyasyncore`) I got the same error message.

imbeacon, 2024-11-04:
Hi @Isac-enerlab,

Thank you for your interest in ThingsBoard IoT Gateway.
You are right, this issue exists. At the moment we have images based on python 3.10 and going to rewrite BACnet connector, to use Bac0 library instead of Bacpypes due to issues like you described and other.

samson0v, 2024-12-03:
Hi @Isac-enerlab and everyone!
We are pleased to inform you that we added a new Async BACnet connector that used bacpypes3 (via PR #1607).
The old BACnet connector is now deprecated and can only be enabled manually.
Please, run the latest version of the gateway via the master branch and let us know about the results.

imbeacon, 2024-12-11:
This issue closed, because relates to outdated implementation of connector, please try the latest version (3.6.2), and feel free to open a new issue if you discover some misbehaviour.

closed 2024-12-11 by imbeacon