thingsboard/thingsboard-gateway #1544: [HELP]  Incorrect padding

pro100805, 2024-09-27:
Good afternoon, after some time the Gateway stops communicating with TB, but continues to poll devices normally and receives data from them.

The content of the /var/log/thingsboard-gateway/storage.log, file looks as follows:  

2024-09-24 14:04:51 - |INFO| - [event_storage_reader.py] - event_storage_reader - delete_read_file - 172 - FileStorage_reader -- Cleanup old data file: ./data/data_1727171359088.txt!
2024-09-24 16:22:22 - |INFO| - [event_storage_reader.py] - event_storage_reader - delete_read_file - 172 - FileStorage_reader -- Cleanup old data file: ./data/data_1727179491161.txt!
2024-09-24 18:36:48 - |INFO| - [event_storage_reader.py] - event_storage_reader - delete_read_file - 172 - FileStorage_reader -- Cleanup old data file: ./data/data_1727187742490.txt!
2024-09-24 20:54:27 - |INFO| - [event_storage_reader.py] - event_storage_reader - delete_read_file - 172 - FileStorage_reader -- Cleanup old data file: ./data/data_1727195808189.txt!
2024-09-24 23:10:36 - |INFO| - [event_storage_reader.py] - event_storage_reader - delete_read_file - 172 - FileStorage_reader -- Cleanup old data file: ./data/data_1727204067841.txt!
2024-09-25 01:27:24 - |INFO| - [event_storage_reader.py] - event_storage_reader - delete_read_file - 172 - FileStorage_reader -- Cleanup old data file: ./data/data_1727212236303.txt!
2024-09-25 03:44:39 - |INFO| - [event_storage_reader.py] - event_storage_reader - delete_read_file - 172 - FileStorage_reader -- Cleanup old data file: ./data/data_1727220444094.txt!
2024-09-25 06:02:12 - |INFO| - [event_storage_reader.py] - event_storage_reader - delete_read_file - 172 - FileStorage_reader -- Cleanup old data file: ./data/data_1727228679510.txt!
2024-09-25 08:20:30 - |INFO| - [event_storage_reader.py] - event_storage_reader - delete_read_file - 172 - FileStorage_reader -- Cleanup old data file: ./data/data_1727236932157.txt!
2024-09-25 10:36:32 - |INFO| - [event_storage_reader.py] - event_storage_reader - delete_read_file - 172 - FileStorage_reader -- Cleanup old data file: ./data/data_1727245230584.txt!
2024-09-25 12:52:42 - |INFO| - [event_storage_reader.py] - event_storage_reader - delete_read_file - 172 - FileStorage_reader -- Cleanup old data file: ./data/data_1727253392190.txt!
2024-09-25 15:11:02 - |INFO| - [event_storage_reader.py] - event_storage_reader - delete_read_file - 172 - FileStorage_reader -- Cleanup old data file: ./data/data_1727261562260.txt!
2024-09-25 17:27:56 - |INFO| - [event_storage_reader.py] - event_storage_reader - delete_read_file - 172 - FileStorage_reader -- Cleanup old data file: ./data/data_1727269862624.txt!
2024-09-25 19:43:49 - |INFO| - [event_storage_reader.py] - event_storage_reader - delete_read_file - 172 - FileStorage_reader -- Cleanup old data file: ./data/data_1727278076773.txt!
2024-09-25 21:59:59 - |INFO| - [event_storage_reader.py] - event_storage_reader - delete_read_file - 172 - FileStorage_reader -- Cleanup old data file: ./data/data_1727286228999.txt!
2024-09-26 00:17:57 - |INFO| - [event_storage_reader.py] - event_storage_reader - delete_read_file - 172 - FileStorage_reader -- Cleanup old data file: ./data/data_1727294399461.txt!
2024-09-26 02:35:35 - |INFO| - [event_storage_reader.py] - event_storage_reader - delete_read_file - 172 - FileStorage_reader -- Cleanup old data file: ./data/data_1727302676958.txt!
2024-09-26 04:52:41 - |INFO| - [event_storage_reader.py] - event_storage_reader - delete_read_file - 172 - FileStorage_reader -- Cleanup old data file: ./data/data_1727310935214.txt!
2024-09-26 07:09:04 - |INFO| - [event_storage_reader.py] - event_storage_reader - delete_read_file - 172 - FileStorage_reader -- Cleanup old data file: ./data/data_1727319161474.txt!
2024-09-26 09:26:08 - |INFO| - [event_storage_reader.py] - event_storage_reader - delete_read_file - 172 - FileStorage_reader -- Cleanup old data file: ./data/data_1727327344866.txt!
2024-09-26 11:42:56 - |INFO| - [event_storage_reader.py] - event_storage_reader - delete_read_file - 172 - FileStorage_reader -- Cleanup old data file: ./data/data_1727335568805.txt!
2024-09-26 14:00:32 - |INFO| - [event_storage_reader.py] - event_storage_reader - delete_read_file - 172 - FileStorage_reader -- Cleanup old data file: ./data/data_1727343776134.txt!
2024-09-26 16:16:52 - |INFO| - [event_storage_reader.py] - event_storage_reader - delete_read_file - 172 - FileStorage_reader -- Cleanup old data file: ./data/data_1727352032856.txt!
2024-09-26 18:35:45 - |INFO| - [event_storage_reader.py] - event_storage_reader - delete_read_file - 172 - FileStorage_reader -- Cleanup old data file: ./data/data_1727360212090.txt!
2024-09-26 20:30:31 - |ERROR| - [event_storage_reader.py] - event_storage_reader - read - 57 - Incorrect padding
Traceback (most recent call last):
  File "/usr/local/lib/python3.10/dist-packages/thingsboard_gateway/storage/file/event_storage_reader.py", line 52, in read
    self.current_batch.append(b64decode(line).decode("utf-8"))
  File "/usr/lib/python3.10/base64.py", line 87, in b64decode
    return binascii.a2b_base64(s)
binascii.Error: Incorrect padding
2024-09-26 21:15:25 - |INFO| - [event_storage_reader.py] - event_storage_reader - delete_read_file - 172 - FileStorage_reader -- Cleanup old data file: ./data/data_1727368544979.txt!

The content of the /var/log/thingsboard-gateway/tb_connection.log, file looks as follows:  

2024-09-21 11:49:37 - |ERROR| - [tb_client.py] - tb_client - _on_connect - 240 - Error in on_connect callback: %s
Traceback (most recent call last):
  File "/usr/local/lib/python3.10/dist-packages/thingsboard_gateway/gateway/tb_client.py", line 236, in _on_connect
    if result_code.getName().lower() == "connection rate exceeded":
AttributeError: 'int' object has no attribute 'getName'
2024-09-21 11:49:37 - |INFO| - [tb_gateway_mqtt.py] - tb_gateway_mqtt - gw_subscribe_to_attribute - 184 - Subscribed to *|* with id 1 for device *
2024-09-21 11:50:00 - |INFO| - [tb_gateway_mqtt.py] - tb_gateway_mqtt - gw_unsubscribe - 192 - Unsubscribed from *|*, subscription id '*'
2024-09-21 11:51:58 - |ERROR| - [tb_client.py] - tb_client - connect - 288 - Connection refused. Check ThingsBoard is running.
2024-09-21 11:51:59 - |ERROR| - [tb_client.py] - tb_client - connect - 288 - Connection refused. Check ThingsBoard is running.
2024-09-21 11:52:00 - |ERROR| - [tb_client.py] - tb_client - connect - 288 - Connection refused. Check ThingsBoard is running.
2024-09-21 11:52:01 - |ERROR| - [tb_client.py] - tb_client - connect - 288 - Connection refused. Check ThingsBoard is running.
2024-09-21 11:52:02 - |ERROR| - [tb_client.py] - tb_client - connect - 288 - Connection refused. Check ThingsBoard is running.
2024-09-21 11:52:03 - |ERROR| - [tb_client.py] - tb_client - connect - 288 - Connection refused. Check ThingsBoard is running.
2024-09-21 11:52:04 - |ERROR| - [tb_client.py] - tb_client - connect - 288 - Connection refused. Check ThingsBoard is running.
2024-09-21 11:52:05 - |ERROR| - [tb_client.py] - tb_client - connect - 288 - Connection refused. Check ThingsBoard is running.
2024-09-21 11:52:06 - |ERROR| - [tb_client.py] - tb_client - connect - 288 - Connection refused. Check ThingsBoard is running.
2024-09-21 11:52:07 - |ERROR| - [tb_client.py] - tb_client - connect - 288 - Connection refused. Check ThingsBoard is running.
2024-09-21 11:52:08 - |ERROR| - [tb_client.py] - tb_client - connect - 288 - Connection refused. Check ThingsBoard is running.
2024-09-21 11:52:09 - |ERROR| - [tb_client.py] - tb_client - connect - 288 - Connection refused. Check ThingsBoard is running.
2024-09-21 11:52:10 - |ERROR| - [tb_client.py] - tb_client - connect - 288 - Connection refused. Check ThingsBoard is running.
2024-09-21 11:52:11 - |ERROR| - [tb_client.py] - tb_client - connect - 288 - Connection refused. Check ThingsBoard is running.
2024-09-21 11:52:12 - |ERROR| - [tb_client.py] - tb_client - connect - 288 - Connection refused. Check ThingsBoard is running.
2024-09-21 11:52:13 - |ERROR| - [tb_client.py] - tb_client - connect - 288 - Connection refused. Check ThingsBoard is running.
2024-09-21 11:52:14 - |ERROR| - [tb_client.py] - tb_client - connect - 288 - Connection refused. Check ThingsBoard is running.
2024-09-21 11:52:15 - |ERROR| - [tb_client.py] - tb_client - connect - 288 - Connection refused. Check ThingsBoard is running.
2024-09-21 11:52:16 - |ERROR| - [tb_client.py] - tb_client - connect - 288 - Connection refused. Check ThingsBoard is running.
[... cut ...]
  },
  "grpc": {
    "enabled": false,
    "serverPort": 9595,
    "keepaliveTimeMs": 10001,
    "keepaliveTimeoutMs": 5000,
    "keepalivePermitWithoutCalls": true,
    "maxPingsWithoutData": 0,
    "minTimeBetweenPingsMs": 10000,
    "minPingIntervalWithoutDataMs": 5000
  },
  "connectors": [
    {
      "type": "modbus",
      "name": "Modbus Connector",
      "configuration": "modbus-serial.json"
    }
  ]
}

It's been over 6 days since the last GW restart. Please help to solve this problem.

pro100805, 2024-09-27:
GW 3.5.1

imbeacon, 2024-11-15:
Hi, @pro100805,

Fix for file storage was added in 3.6.1

closed 2024-11-15 by imbeacon