thingsboard/thingsboard-gateway #1545: [HELP] Timeout while waiting for service configuration!, session will use default configuration

devaskim, 2024-09-29:
**Describe the issue**
After upgrading gateway to 3.5.2 I observe slowing down first data push after gateway is started. Seems like even with disabled `remoteConfiguration` option, gateway requests some information from a server (TB CE 3.7.0).
```
2024-09-29 12:18:36 - |INFO| - [tb_gateway_service.py] - tb_gateway_service - __init__ - 246 - Gateway started.
2024-09-29 12:19:31 - |WARNING| - [tb_gateway_mqtt.py] - tb_gateway_mqtt - __on_service_configuration - 320 - Timeout while waiting for service configuration!, session will use default configuration.
```
What should be done on server side or on gateway side to speed up first push and to get rid of the following warning?

**Versions (please complete the following information):**
 - OS: docker compose
 - Thingsboard IoT Gateway version [e.g. 3.5.2]
 - Python version[e.g. 3.9]

imbeacon, 2024-09-30:
Hi @devaskim,

You are right, starting from ThingsBoard 3.8.0 - gateway will be able to get session limitations, like rate limits, maximal payload size, max inflight messages, it takes some time. Such as this functionality not present in previous versions - this request fails with timeout and uses default or manually configured rate limits. This request sends once, right after connect, so it won’t affect much data processing after connecting to ThingsBoard.

devaskim, 2024-09-30:
@imbeacon Thanks for quick response

closed 2024-09-30 by devaskim