thingsboard/thingsboard-gateway #1547: [BUG] Device type (profile) is not applied

devaskim, 2024-09-29:
**Describe the bug**

This issue is reproduced both if profile (type) is previously created on a server or not. On previous versions on gateway/server new profile is automatically created.

While debugging with PyCharm debugger, I see that device type is properly resolved by JSON converter and data packet is passed to gateway core.

JSON Converter configuration:
```
    {
      "topicFilter": "ESA/Toilet/Mandosi/Boys/D1",
      "converter": {
        "type": "json",
        "deviceNameTopicExpression": "(?<=ESA/Toilet/)(.*?)/(.*?)/D\\d{1,}",
        "deviceTypeTopicExpression": "Toilets Controller",
        "sendDataOnlyOnChange": false,
        "timeout": 60000,
        "attributes": [],
        "timeseries": "*"
      }
    }
```

[image]

**Connector name (If bug in the some connector):**
MQTT

**Versions (please complete the following information):**
 - Deployment: docker compose
 - Thingsboard IoT Gateway version [e.g. 3.5.2]
 - Thingsboard Server: CE 3.7.0
 - Python version[e.g. 3.7]

khoahoangqenergy, 2024-10-16:
We are facing the same issue, device profile is not getting created.

devaskim, 2024-11-22:
@imbeacon @samson0v Any thoughts?

closed 2025-05-08 by imbeacon