thingsboard/thingsboard-gateway #1548: The OPC UA connector does not read the folder tree created by the M251 PLC correctly.

aramosf1966, 2024-10-02:
I need to connect the Thingsboard platform to the Schneider M251 PLC using the OPC UA protocol that is supported by both the Client-Server. The OPCUA connector is already configured and in the connector mapping part it only reads the tree folders and does not read the variables and therefore there is no telemetry.
I don't realize where the error is. Could you help me??
Thank you....
Create  "mapping": [
      {
        "deviceNodePattern": "Root\\.Objects\\.Device1",
        "deviceNamePattern": "Device ${Root\\.Objects\\.Device1\\.serialNumber}",
        "attributes": [
          {
            "key": "CertificateNumber",
            "path": "${ns=2;i=5}"
          }
        ],
        "timeseries": [
          {
            "key": "temperature °C",
            "path": "${Root\\.Objects\\.Device1\\.TemperatureAndHumiditySensor\\.Temperature}"

JavierNR, 2024-10-10:
@aramosf1966 Probably #1555 will help you, as I have these PLC models connected. 
Appears that you have malformed config JSON.

Check my screenshots on the issue.

aramosf1966, 2024-10-10:
Hello..I just saw your suggestion to find the error that I must have in the json of the UA connector to connect the Schneider M251 PLC to the Thinsboard Platform. I have not been able to solve it yet. I will try your suggestion next Monday. I will send you my json file of the connector and perhaps you can see where I have the error. Thank you very much for the help.

{
  "server": {
    "name": "OPC-UA Default Server",
    "url": "opc.tcp://<my-ip>:4840/freeopcua/server/",
    "timeoutInMillis": 5000,
    "scanPeriodInMillis": 5000,
    "disableSubscriptions": false,
    "subCheckPeriodInMillis": 100,
    "showMap": false,
    "security": "none",
    "identity": {
      "type": "anonymous"
    },
    "mapping": [
      {
        "deviceNodePattern": "Root\\.Objects\\.M241-M251 data",
        "deviceNamePattern": "PLC",
        "timeseries": [
          {
            "key": "temperature",
            "path": "${GVL_INP.iSondaEntradaHorno}"
          }
        ]
      }
    ]
  },
  "logLevel": "DEBUG",
  "name": "Planta",
  "enableRemoteLogging": true,
  "id": "6b6beaa5-7178-4e89-b5fa-6829ae3012a1"

aramosf1966, 2024-10-30:
thanks so much!! all it's working now @JavierNR

closed 2024-10-30 by aramosf1966