thingsboard/thingsboard-gateway #1549: Receive unnecessary messages from IoT Gateway Thingboards Docker, MQTT

duybaoks, 2024-10-03:
Hi every one. 

I have installed the IoT Gateway on an AWS EC2 Linux instance using Docker Compose, based on the docker-compose.yml file.

The installation was successful, and I have configured it to receive data from the IoT Gateway Docker. However, even after turning off statictic.py to stop sending telemetry data in the tb-gateway container and restarting it, I still see many messages like the ones in the attached image in the rule chain node. I only want to get the messenger necessary the same in the configuration broker .

I followed the installation instructions exactly. I have the JSON configuration broker below related to the broken MQTT connection in the IoT gateway.

 
"broker": {
"name": "xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx",
"host": "eu1.cloud.thethings.network",
"port": 1883,
"version": 4,
"clientId": "xxxxxxxxx",
"maxNumberOfWorkers": 100,
"maxMessageNumberPerWorker": 10,
"security": {
"type": "basic",
"username": "xxxxxxx@xxxxxxxx",
"password": "xxxxx.xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx"
}
},
"dataMapping": [
{
"topicFilter": "v3/xxxxxx@xxxxxx/devices/+/up",
"subscriptionQos": 0,
"converter": {
"type": "json",
"deviceInfo": {
"deviceNameExpressionSource": "message",
"deviceNameExpression": "${end_device_ids.device_id}",
"deviceProfileExpressionSource": "constant",
"deviceProfileExpression": "SensorProfile"
},
"attributes": [
{
"type": "json",
"key": "device_attributes",
"value": "${end_device_ids}"
},
{
"type": "json",
"key": "version_attributes",
"value": "${version_ids}"
}
],
"timeseries": [
{
"type": "json",
"key": "payload",
"value": "${uplink_message.decoded_payload}"
},
{
"type": "json",
"key": "metadata",
"value": "${uplink_message.rx_metadata}"
}
],
"telemetryFilter": [
{
"key": "storage_ERRORS_COUNT",
"condition": "False"
},
{
"key": "ALL_ERRORS_COUNT",
"condition": "False"
}
]
}
}
],
"requestsMapping": {
"connectRequests": [
{
"topicFilter": "v3/tb-community-test@ttn/devices/+/up",
"deviceInfo": {
"deviceNameExpressionSource": "message",
"deviceNameExpression": "${end_device_ids.device_id}",
"deviceProfileExpressionSource": "constant",
"deviceProfileExpression": "SensorProfile"
}
}
]
},
"logLevel": "NONE",
"name": "xxxxxxxxxxxxxxxxxxxxxxxxxxxxxx",
"enableRemoteLogging": false,
"id": "xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx"
}
 
Please help me resolve this issue?
I use ThingsBoard versions v3.7.0 (Jun 17, 2024)

Best Regard
Thank you 
 
[image]
[image]
[image]

duybaoks, 2024-10-11:
Have any one encountered this situation? Please help me

imbeacon, 2025-04-11:
Hi @duybaoks,

In the gateway 3.7.3 we have added ability to completely disable statistics - just set enable:False in tb_gateway.json (statistics) section. According to error count - it is a service messages and there is no option to disable them.

closed 2025-04-16 by imbeacon