thingsboard/thingsboard-gateway #1551: [BUG] BacNet connector broken in version 3.5.2, 3.5.3, 3.5.3.1

jurkov, 2024-10-04:
**Describe the bug**
Starting a newly created iot-gateway with docker or native python throws the same error since version 3.5.2.
Version 3.5.1 works fine.

**Connector name (If bug in the some connector):**
BacNET Connector

**Error traceback (If available):**
```
  File "C:\projects\thingsboard-gateway-native\.venv\Lib\site-packages\thingsboard_gateway\connectors\bacnet\bacnet_uplink_converter.py", line 53, in convert
    count=len(dict_result["attributes"]))
              ~~~~~~~~~~~^^^^^^^^^^^^^^
TypeError: string indices must be integers, not 'str'
```

**Versions (please complete the following information):**
 - OS: Windows 11 or Docker Image OS
 - Thingsboard IoT Gateway version 3.5.2, 3.5.3, 3.5.3.1
 - Python version 3.11, Python version 3.12 (installed pyasyncore by hand)

jurkov, 2024-10-10:
A workaround would be to downgrade to 3.5.1. I fixed line 174 in tb_client.py and replaced `'DEBUG'` with `10`. (To be able to use MQTTS).

sanjayN4497, 2024-11-08:
Hi @imbeacon/ @jurkov,

i have downgraded gateway with version 3.5.1 and replaced 174 line, but still same error reoccuring.

Line 174:  if self.__logger.isEnabledFor('10'):

Error : convert
    count=len(dict_result["attributes"]))
    
    
[image]

jurkov, 2024-11-11:
Try tp replace the string `'10'` with the integer `10`.

sanjayN4497, 2024-11-11:
Hi @jurkov , 

as you can see below in below snap i change '10' with integer 10 but issue remain same as below:

[image]

imbeacon, 2024-12-11:
This issue closed, because relates to outdated implementation of connector, please try the latest version (3.6.2), and feel free to open a new issue if you discover some misbehaviour.

closed 2024-12-11 by imbeacon

jurkov, 2025-01-08:
I can confirm, that 3.6.2 works much better. I had to remove the port from the address of every device or use _altResponsesAddresses_ make it work.
Example:
`      "altResponsesAddresses": [
        "192.168.x.y:47808"
      ]`