thingsboard/thingsboard-gateway #1564: [HELP]How to Automatically Monitor and Update Gateway Versions

DuXiaoChuang, 2024-10-21:
**Describe the issue**
How to Automatically Monitor and Update Gateway Versions，I see the updated URL in the code, but the URL cannot be opened, the error is as shown below.  How should I build my own update website, please guide me, thank you

[image]

yhliyr, 2024-10-21:
this url can only be request via POST method(as you can see in the TBUpdater.check_for_new_version() method in tb_updater.py), not via GET, such as by accessing the url in a browser

DuXiaoChuang, 2024-10-21:
> this url can only be request via POST method(as you can see in the TBUpdater.check_for_new_version() method in tb_updater.py), not via GET, such as by accessing the url in a browser

How should I build my own update website, please guide me, thank you
[image]

yhliyr, 2024-10-22:
From what I understand based on the code, the update process for Thingsboard-gateway is initiated by user sending RPC command from the platform(Thingsboard), the gateway receives the command and use 'pip install' to install the new version of the thingsboard-gateway python package and then reloads itself afterwards
So it depends on what kind of 'my own update website' you want

about the rpc command: https://github.com/thingsboard/thingsboard-gateway/issues/1459#issuecomment-2257599797