thingsboard/thingsboard-gateway #1565: [HELP]  Bacnet connector not working. Devices not connecting.

KIOfmorenocruz, 2024-10-21:
**Describe the issue**
I can’t make the BACnet connector work. ThingsBoard IoT gateway is running on a revolution Pi. We have a network full of devices and we only receive the response from one (in the screenshot you can see the IP) and none of them is able to receive the proper confirmation and to pull the telemetry. Even writing that IP in the devices list of the bacnet.json. The other devices do not show up.
Is there something in the connector config json wrong? is it a known bug?

**Configuration**

{
  "thingsboard": {
    "host": "abc.xyz",
    "port": 8883,
    "remoteShell": true,
    "remoteConfiguration": true,
    "statistics": {
      "enable": true,
      "statsSendPeriodInSeconds": 3600,
      "commands": []
    },
    "deviceFiltering": {
      "enable": false,
      "filterFile": "list.json"
    },
    "maxPayloadSizeBytes": 8196,
    "minPackSendDelayMS": 50,
    "minPackSizeToSend": 500,
    "checkConnectorsConfigurationInSeconds": 60,
    "handleDeviceRenaming": true,
    "security": {
      "type": "accessToken",
      "accessToken": "xxx",
      "caCert": "ca.crt"
    },
    "qos": 1,
    "checkingDeviceActivity": {
      "checkDeviceInactivity": false,
      "inactivityTimeoutSeconds": 200,
      "inactivityCheckPeriodSeconds": 500
    },
    "ts": 1729001934166,
    "rateLimits": "DEFAULT_TELEMETRY_RATE_LIMIT",
    "dpRateLimits": "DEFAULT_TELEMETRY_DP_RATE_LIMIT",
    "messagesRateLimits": "DEFAULT_MESSAGES_RATE_LIMIT",
    "deviceMessagesRateLimits": "DEFAULT_MESSAGES_RATE_LIMIT",
    "deviceRateLimits": "DEFAULT_TELEMETRY_RATE_LIMIT",
    "deviceDpRateLimits": "DEFAULT_TELEMETRY_DP_RATE_LIMIT"
  },
  "grpc": {
    "enabled": false,
    "serverPort": 9595,
    "keepaliveTimeMs": 10000,
    "keepaliveTimeoutMs": 5000,
    "keepalivePermitWithoutCalls": true,
    "maxPingsWithoutData": 0,
    "minTimeBetweenPingsMs": 10000,
    "minPingIntervalWithoutDataMs": 5000,
    "keepAliveTimeMs": 10000,
    "keepAliveTimeoutMs": 5000,
    "ts": 1729424816977
  },
  "logs": { xxx },
  "storage": {
    "type": "file",
    "read_records_count": 100,
    "max_records_count": 10000,
    "ts": 1729001934166,
    "data_folder_path": "./data/",
    "max_file_count": 10,
    "max_read_records_count": 10,
    "max_records_per_file": 10000,
    "data_file_path": "./data/data.db",
    "messages_ttl_check_in_hours": 1,
    "messages_ttl_in_days": 7
  },
  "mode": "basic"
}

**Connector name (If you need help with some connector/converter):**
Bacnet
[image]

**Error traceback (If it was raised):**
[image]

**Versions (please complete the following information):**
Latest

imbeacon, 2024-11-04:
Hi @KIOfmorenocruz,

Thank you for your interest in ThingsBoard IoT Gateway.
In your connector configuration - I cannot see any related to the data collection configs sections, like timeseries, attributes, etc. You can use the following config as a reference to configure devices that you want connect to - https://github.com/thingsboard/thingsboard-gateway/blob/master/thingsboard_gateway/config/bacnet.json

samson0v, 2024-12-03:
Hi @KIOfmorenocruz and everyone!
We are pleased to inform you that we added a new Async BACnet connector that used bacpypes3 (via PR #1607).
The old BACnet connector is now deprecated and can only be enabled manually.
Please, run the latest version of the gateway via the master branch and let us know about the results.

imbeacon, 2024-12-11:
This issue closed, because relates to outdated implementation of connector, please try the latest version (3.6.2), and feel free to open a new issue if you discover some misbehaviour.

closed 2024-12-11 by imbeacon