thingsboard/thingsboard-gateway #1574: [HELP] Set file path to config files

andersenthomas98, 2024-10-29:
**Describe the issue**
Hi! Is it possible to specify the file path to the config files (config/tb_gateway, etc...) necessary to run the thingsboard gateway?

imbeacon, 2024-11-04:
Hi @andersenthomas98,

Unfortunately, there is no such option for now, you can do this only by modifying thingsboard_gateway/tb_gateway.py file of your gateway. Probably we will add ability to use environment variable to configure the path to config folder in future releases.

imbeacon, 2024-11-04:
Added this functionality in https://github.com/thingsboard/thingsboard-gateway/commit/f8ace4649d957e97e086cef9c376a7760aa1ef09, you can use the version from the master branch or wait for release, environmental variable for config folder location is:
```
TB_GW_CONFIG_DIR
```

closed 2024-11-05 by andersenthomas98