thingsboard/thingsboard-gateway #1575: [HELP]

tangchengxiang666, 2024-10-31:
**Describe the issue**
The Modbus RPC call failed

**Configuration** (Attach your configuration file)
**Notate: Remove Access token from file if you want to attach a tb_gateway.yaml**

**Connector name (If you need help with some connector/converter):**
MODBUS

**Error traceback (If it was raised):**

The 空调遥控器1 connector should be modbus1
The 空调遥控器2 connector should be modbus1
[image]
This causes me to have the following anomaly
```
2024-10-31 16:56:34 - |ERROR| - [tb_logger.py] - tb_logger - exception - 161 - Error during RPC handling: %s
Traceback (most recent call last):
  File "/home/tcx/Documents/code/github/thingsboard-gateway-release-3.5.3/thingsboard_gateway/connectors/modbus/modbus_connector.py", line 701, in server_side_rpc_handler
    device = ModbusConnector.__get_device_by_name(server_rpc_request[DEVICE_SECTION_PARAMETER],
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/tcx/Documents/code/github/thingsboard-gateway-release-3.5.3/thingsboard_gateway/connectors/modbus/modbus_connector.py", line 837, in __get_device_by_name
    return tuple(filter(lambda slave: slave.device_name == device_name, devices))[0]
           ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^
IndexError: tuple index out of range
```

**Versions (please complete the following information):**
 - OS: [Ubuntu 24.04]
 - Thingsboard IoT Gateway version [3.5.3]
 - Python version[3.11]

pro100805, 2024-11-17:
Nov 17 11:23:51 us-tb-gt-193 python3[7951]:   File "/var/lib/thingsboard_gateway/venv/lib/python3.12/site-packages/thingsboard_gateway/gateway/tb_gateway_service.py", line 995, in __connect_with_connectors
Nov 17 11:23:51 us-tb-gt-193 python3[7951]:     and len(connector_config[CONFIG_SECTION_PARAMETER][config].keys()) > 3) or \
Nov 17 11:23:51 us-tb-gt-193 python3[7951]:             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
Nov 17 11:23:51 us-tb-gt-193 python3[7951]: AttributeError: 'str' object has no attribute 'keys'
Nov 17 11:23:51 us-tb-gt-193 python3[7951]: 2024-11-17 11:23:51 - |INFO| - [tb_client.py] - tb_client - connect - 322 - Connecting to ThingsBoard...
Nov 17 11:23:51 us-tb-gt-193 python3[7951]: 2024-11-17 11:23:51 - |INFO| - [tb_client.py] - tb_client - connect - 326 - Sending connect to platform...
Nov 17 11:23:51 us-tb-gt-193 python3[7951]: 2024-11-17 11:23:51 - |INFO| - [tb_client.py] - tb_client - connect - 329 - Connect msg sent to platform.
Nov 17 11:23:51 us-tb-gt-193 python3[7951]: 2024-11-17 11:23:51 - |INFO| - [tb_device_mqtt.py] - tb_device_mqtt - _on_connect - 436 - MQTT client <paho.mqtt.client.Client object at 0x7ea9c78ff950> - Connected!
Nov 17 11:23:52 us-tb-gt-193 python3[7951]: 2024-11-17 11:23:52 - |INFO| - [tb_device_mqtt.py] - tb_device_mqtt - on_service_configuration - 738 - Service configuration was successfully retrieved and applied.
Nov 17 11:23:52 us-tb-gt-193 python3[7951]: 2024-11-17 11:23:52 - |INFO| - [tb_device_mqtt.py] - tb_device_mqtt - on_service_configuration - 739 - Current limits: {'rateLimit': {'messages': None, 'telemetryMessages': None, 'telemetryDataPoints': None}, 'maxPayloadSize': 65536, 'maxInflightMessages': 100}

OS 24.04
TB 3.8.1
GW 3.6.1
p 3.12

imbeacon, 2025-07-08:
Hi,

The new version is released. Please try to use it, feel free to open new issue if you discover something.

closed 2025-07-08 by imbeacon