thingsboard/thingsboard-gateway #1577: [HELP] Сommunication failure between the gateway and TB after 6 days of normal communication

pro100805, 2024-11-01:
Everything works fine for 6 days and then TB stops seeing GW. The GW continues to poll devices and receive telemetry normally

gateway polls devices using serial port Modbus RTU protocol

restarting the gateway or TB does not bring any result. After a complete reboot of the system the situation repeats, everything works fine for about 6 days

the same server and gateway using communication with Modbus TCP/IP devices works without problems

=================================================
tb_gateway.json

  "thingsboard": {
    "host": "************",
    "port": 1883,
    "remoteShell": false,
    "remoteConfiguration": true,
    "statistics": {
      "enable": true,
      "statsSendPeriodInSeconds": 60,
      "customStatsSendPeriodInSeconds": 3600
    },
    "deviceFiltering": {
      "enable": false,
      "filterFile": "list.json"
    },
    "maxPayloadSizeBytes": 8196,
    "minPackSendDelayMS": 50,
    "minPackSizeToSend": 500,
    "checkConnectorsConfigurationInSeconds": 60,
    "handleDeviceRenaming": true,
    "security": {
      "type": "accessToken",
      "accessToken": "**********************"
    },
    "qos": 1,
    "checkingDeviceActivity": {
      "checkDeviceInactivity": false,
      "inactivityTimeoutSeconds": 200,
      "inactivityCheckPeriodSeconds": 500
    }
  },
  "storage": {
    "type": "memory",
    "read_records_count": 100,
    "max_records_count": 100000
  },
  "grpc": {
    "enabled": false,
    "serverPort": 9595,
    "keepaliveTimeMs": 10001,
    "keepaliveTimeoutMs": 5000,
    "keepalivePermitWithoutCalls": true,
    "maxPingsWithoutData": 0,
    "minTimeBetweenPingsMs": 10000,
    "minPingIntervalWithoutDataMs": 5000
  },
  "connectors": [
    {
      "type": "modbus",
      "name": "Modbus Connector",
      "configuration": "modbus-serial.json"
    }
  ]
}
=================================================
tb_connection.log

2024-10-23 19:52:57 - |ERROR| - [tb_client.py] - tb_client - _on_connect - 240 - Error in on_connect callback: %s
Traceback (most recent call last):
  File "/usr/local/lib/python3.10/dist-packages/thingsboard_gateway/gateway/tb_client.py", line 236, in _on_connect
    if result_code.getName().lower() == "connection rate exceeded":
AttributeError: 'int' object has no attribute 'getName'
2024-10-23 19:52:57 - |INFO| - [tb_gateway_mqtt.py] - tb_gateway_mqtt - gw_subscribe_to_attribute - 184 - Subscribed to *|* with id 1 for device *
2024-10-26 18:20:46 - |ERROR| - [tb_client.py] - tb_client - _on_connect - 240 - Error in on_connect callback: %s
Traceback (most recent call last):
  File "/usr/local/lib/python3.10/dist-packages/thingsboard_gateway/gateway/tb_client.py", line 236, in _on_connect
    if result_code.getName().lower() == "connection rate exceeded":
AttributeError: 'int' object has no attribute 'getName'
2024-10-26 18:20:46 - |INFO| - [tb_gateway_mqtt.py] - tb_gateway_mqtt - gw_subscribe_to_attribute - 184 - Subscribed to *|* with id 1 for device *
======================================================

 - OS: Ubuntu 22.04
 -  Thingsboard version 3.7.0
 - Thingsboard IoT Gateway version 3.5.1
 - Python version 3.10

imbeacon, 2024-11-04:
Hi @pro100805,

Thank you for your interest in ThingsBoard IoT Gateway.
Could you check that you are using the latest version of tb-mqtt-client? 
You can do this using the following command:
```
pip show tb-mqtt-client
```

Version should be 1.10.8 at least, if it is older you can update it manually, using the following command:

```
pip install —upgrade tb-mqtt-client
```

Please try to update the client, restart and then check after the same period of time, it should be solved.

pro100805, 2024-11-04:
thanks for the quick response
version was 1.9.9, updated to see what the result will be 
there is a small typo in the command, one “-” is missing.
pip install --upgrade tb-mqtt-client
You continue to exceed every expectation that we set. Great job.

pro100805, 2024-11-10:
Good day, unfortunately the situation is the same again, GW continues to receive data normally, but the connection with TB is broken. 5 days of normal operation and then this situation. what additional information can I give you to sort this out?

pro100805, 2024-11-10:
[image]

pro100805, 2024-11-10:
[image]

apachler, 2025-05-12:
I see also an issue with the connection to Thingsboard. If the MQTT transport is not available for some reason, it seems the GW does not reconnect properly if the IP of the MQTT transport changes. This is odd because i have configured a hostname in GW for the MQTT transport.