thingsboard/thingsboard-gateway #1578: [BUG] Install the gateway 3.5.3 using apt

pro100805, 2024-11-02:
the following information appears during a clean installation 

@test:~$ sudo apt install ./python3-thingsboard-gateway.deb -y
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Note, selecting 'python3-thingsboard-gateway' instead of './python3-thingsboard-gateway.deb'
python3-thingsboard-gateway is already the newest version (3.5.3).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
1 not fully installed or removed.
After this operation, 0 B of additional disk space will be used.
Setting up python3-thingsboard-gateway (3.5.3) ...
Running py3compile for python3-thingsboard-gateway...
Checking if /var/lib/thingsboard_gateway/ exists...
/var/lib/thingsboard_gateway/ already exists.
Copying extensions to /var/lib/thingsboard_gateway/...
cp: cannot stat '/etc/thingsboard-gateway/extensions': No such file or directory
dpkg: error processing package python3-thingsboard-gateway (--configure):
 installed python3-thingsboard-gateway package post-installation script subprocess returned error exit status 1
Errors were encountered while processing:
 python3-thingsboard-gateway
E: Sub-process /usr/bin/dpkg returned an error code (1)
===============================
@test:~$ cd /etc/thingsboard-gateway/
@test:/etc/thingsboard-gateway$ ls
config  extensions.tar.gz
@test:/etc/thingsboard-gateway$

 - OS: Ubuntu Server 22.04.5
 - Thingsboard version 3.8.1
 - Thingsboard IoT Gateway version 3.5.3
 - Python version[e.g. 3.10

pro100805, 2024-12-11:
Thank you for your reply I am currently trying version 3.6.1
but I am having trouble applying RPC requests

MaxSchoeman, 2025-02-02:
Hey @pro100805 was this resolved?

pro100805, 2025-02-16:
I am currently testing version 3.7 with no problems with telemetry transmission
 but the RPC Get function does not work (Set works)