thingsboard/thingsboard-gateway #1579: [BUG] some messages failed to report

ppmll, 2024-11-02:
**Describe the issue**
I think this is a bug but I am not sure. 
When more than one timestamped messages sent to the storage for the same device, the gateway only reports the last one to thingsboard, 
I've made some debug and discovered that a method called _split_message is called with all the messages and just one is returned before it is beeing publish to mqtt.

**Configuration** (Attach your configuration file)
Not needed

**Connector name (If you need help with some connector/converter):**
I am using a simple custom connector that reads a specific file, each line of that file has a specific timestamp, it is then sending to the storage using gateway.send_to_storage

**Error traceback (If it was raised):**
Not raised

**Versions (please complete the following information):**
 - OS:  Ubuntu 20.04
 - Thingsboard IoT Gateway version 3.6
 - Python version 3.12

imbeacon, 2024-11-04:
Hi @ppmll,

You are particularly right, the gateway collects and send only last value from the converted data, it is done in the same way as implemented in ThingsBoard itself - data with the same ts cannot exists. To avoid this situation - you can add ts to your data and save ConvertedData with TelemetryEntry and ts, this will avoid overwriting of the data.

ppmll, 2024-11-04:
Hi @imbeacon, I am sorry, maybe I did not explain correctly, what I mean is that if I send 3 consecutive messages with different timestamps only the last one is sent to thingsboard, please see the folllowing example:
Input of the _split_message method:
[image]

Thanks.

imbeacon, 2024-11-04:
I see and able to reproduce, let you know after fixing this point.

imbeacon, 2024-11-05:
Hi @ppmll,

I have made some fix for this data loss, please try new version of the sdk - https://github.com/thingsboard/thingsboard-python-client-sdk/releases/tag/1.10.10, or the gateway from the master branch.

ppmll, 2024-11-05:
It seems ok now, thank you.

closed 2024-11-05 by ppmll