thingsboard/thingsboard-gateway #1584: [HELP] Establish MQTT connection to thingsboard cloud using x509 certificates

andersenthomas98, 2024-11-05:
**Describe the issue**
Hi, I am trying to establish an MQTT connection to thingsboard cloud using X509 certificates. I have tried following the configuration guide here: https://thingsboard.io/docs/iot-gateway/configuration/.

I have uploaded the X.509 certificate in pem format in Thingsboard Cloud (Devices --> My device ( is gateway ) --> Manage Credentials --> X.509 certificate). The correct certificates are added to my environment. 

**Configuration**
My docker-compose.yaml file looks like this: 
```yaml
services:
  # ThingsBoard IoT Gateway Service Configuration
  tb-gateway:
    image: thingsboard/tb-gateway:3.5.3
    container_name: tb-gateway
    restart: always

    # Ports bindings - required by some connectors
    ports:
      - "8883:8883"

    # Necessary mapping for Linux
    extra_hosts:
      - "host.docker.internal:host-gateway"

    # Environment variables
    environment:
      - TB_GW_HOST=mqtt.thingsboard.cloud
      - TB_GW_PORT=8883
      - TB_GW_CA_CERT=credentials/ca-root.pem
      - TB_GW_PRIVATE_KEY=credentials/key.pem
      - TB_GW_CERTIFICATE=credentials/cert.pem

    # Volumes bind
    volumes:
      - ./credentials:/thingsboard_gateway/config/credentials
      - tb-gw-config:/thingsboard_gateway/config
      - tb-gw-logs:/thingsboard_gateway/logs
      - tb-gw-extensions:/thingsboard_gateway/extensions

# Volumes declaration for configurations, extensions and configuration
volumes:
  tb-gw-config:
    name: tb-gw-config
  tb-gw-logs:
    name: tb-gw-logs
  tb-gw-extensions:
    name: tb-gw-extensions
```
**Error traceback (If it was raised):**
This is printed to the console:
```
Attaching to tb-gateway
tb-gateway  | Cannot load GRPC connector!
tb-gateway  | 2024-11-05 14:50:38 - |INFO| - [tb_gateway_service.py] - tb_gateway_service - __init__ - 175 - Gateway starting...
tb-gateway  | 2024-11-05 14:50:38 - |INFO| - [tb_gateway_service.py] - tb_gateway_service - __init__ - 180 - ThingsBoard IoT gateway version: 3.5.3
tb-gateway  | 2024-11-05 14:50:38 - |WARNING| - [tb_gateway_service.py] - tb_gateway_service - _load_connectors - 932 - Connectors - not found!
tb-gateway  | /thingsboard_gateway/tb_utility/tb_utility.py:283: CryptographyDeprecationWarning: Properties that return a naïve datetime object have been deprecated. Please switch to not_valid_after_utc.
tb-gateway  |   if cert_detail.not_valid_after - datetime.datetime.now() <= datetime.timedelta(days=days_left):
tb-gateway  | 2024-11-05 14:50:38 - |INFO| - [tb_client.py] - tb_client - connect - 313 - Connecting to ThingsBoard...
tb-gateway  | 2024-11-05 14:50:40 - |INFO| - [tb_client.py] - tb_client - connect - 313 - Connecting to ThingsBoard...
tb-gateway  | 2024-11-05 14:50:41 - |ERROR| - [tb_device_mqtt.py] - tb_device_mqtt - _on_connect - 409 - connection FAIL with error Not authorized Not authorized
tb-gateway  | 2024-11-05 14:50:41 - |WARNING| - [tb_device_mqtt.py] - tb_device_mqtt - _on_disconnect - 387 - MQTT client was disconnected with reason code 2 (A network protocol error occurred when communicating with the broker.)
tb-gateway  | 2024-11-05 14:50:42 - |INFO| - [tb_client.py] - tb_client - connect - 313 - Connecting to ThingsBoard...
tb-gateway  | 2024-11-05 14:50:42 - |ERROR| - [tb_device_mqtt.py] - tb_device_mqtt - _on_connect - 409 - connection FAIL with error Not authorized Not authorized
tb-gateway  | 2024-11-05 14:50:42 - |WARNING| - [tb_device_mqtt.py] - tb_device_mqtt - _on_disconnect - 387 - MQTT client was disconnected with reason code 2 (A network protocol error occurred when communicating with the broker.)
tb-gateway  | 2024-11-05 14:50:43 - |INFO| - [tb_client.py] - tb_client - connect - 313 - Connecting to ThingsBoard...
tb-gateway  | 2024-11-05 14:50:44 - |ERROR| - [tb_device_mqtt.py] - tb_device_mqtt - _on_connect - 409 - connection FAIL with error Not authorized Not authorized
tb-gateway  | 2024-11-05 14:50:44 - |WARNING| - [tb_device_mqtt.py] - tb_device_mqtt - _on_disconnect - 387 - MQTT client was disconnected with reason code 2 (A network protocol error occurred when communicating with the broker.)
```

**Versions (please complete the following information):**
 - OS: Ubuntu 24.04 LTS
 - Thingsboard IoT Gateway version: 3.5.3
 - Docker version: 27.3.1

samson0v, 2024-11-06:
Hi @andersenthomas98, thanks for your interest in ThingsBoard IoT Gateway!
Please, use `TB_GW_CERT` instead of `TB_GW_CERTIFICATE` env variable to avoid this situation.
Like this:
[image]

andersenthomas98, 2024-11-06:
Ah thank you, looks like it is working now! :)

closed 2024-11-06 by andersenthomas98