thingsboard/thingsboard-gateway #1585: [HELP] I launch the IoT Gateway but getting connection to (host.docker.internal, 5021) failed, time out

KashifAhmd, 2024-11-06:
**Describe the issue**
I am trying to Launch the Iot Gateway but getting connection failed issue.

**Configuration** (docker-compose.yml)
```
services:
  # ThingsBoard IoT Gateway Service Configuration
  tb-gateway:
    image: thingsboard/tb-gateway:latest
    container_name: tb-gateway
    restart: always

    # Necessary mapping for Linux
    extra_hosts:
      - "host.docker.internal:host-gateway"

    # Environment variables
    environment:
      - host=myHost.io
      - port=1883
      - accessToken=myAccessToken

```

**The response that I was getting**
```
kashif@XIOPL-DEV-LAP-04:~/my_project$ docker-compose up
[+] Running 1/0
 ✔ Container tb-gateway  Created                                                                                                                                                                              0.0s 
Attaching to tb-gateway
tb-gateway  | Cannot load GRPC connector!
tb-gateway  | 2024-11-06 09:03:45 - |INFO| - [tb_gateway_service.py] - tb_gateway_service - __init__ - 175 - Gateway starting...
tb-gateway  | 2024-11-06 09:03:45 - |INFO| - [tb_gateway_service.py] - tb_gateway_service - __init__ - 180 - ThingsBoard IoT gateway version: 3.5.3
tb-gateway  | 2024-11-06 09:03:45 - |INFO| - [tb_loader.py] - tb_loader - import_module - 68 - Import ModbusConnector from /thingsboard_gateway/connectors/modbus.
tb-gateway  | 2024-11-06 09:03:46 - |INFO| - [tb_client.py] - tb_client - connect - 313 - Connecting to ThingsBoard...
tb-gateway  | 2024-11-06 09:03:48 - |INFO| - [tb_client.py] - tb_client - connect - 313 - Connecting to ThingsBoard...
tb-gateway  | 2024-11-06 09:03:48 - |INFO| - [tb_device_mqtt.py] - tb_device_mqtt - _on_connect - 396 - MQTT client <paho.mqtt.client.Client object at 0x7f9be86a2ad0> - Connected!
tb-gateway  | 2024-11-06 09:03:48 - |INFO| - [tb_device_mqtt.py] - tb_device_mqtt - on_service_configuration - 683 - Service configuration was successfully retrieved and applied.
tb-gateway  | 2024-11-06 09:03:48 - |INFO| - [tb_device_mqtt.py] - tb_device_mqtt - on_service_configuration - 684 - Current limits: {'rateLimit': {'messages': None, 'telemetryMessages': None, 'telemetryDataPoints': None}, 'maxPayloadSize': 65536, 'maxInflightMessages': 100}
tb-gateway  | 2024-11-06 09:03:49 - |ERROR| - [tcp.py] - tcp - connect - 187 - Connection to (host.docker.internal, 5021) failed: timed out
tb-gateway  | 2024-11-06 09:03:52 - |ERROR| - [tcp.py] - tcp - connect - 187 - Connection to (host.docker.internal, 5022) failed: timed out
tb-gateway  | 2024-11-06 09:03:53 - |INFO| - [tb_gateway_service.py] - tb_gateway_service - __init__ - 246 - Gateway started.
tb-gateway  | 2024-11-06 09:03:53 - |INFO| - [tb_gateway_mqtt.py] - tb_gateway_mqtt - gw_subscribe_to_attribute - 255 - Subscribed to *|* with id 1 for device *
tb-gateway  | 2024-11-06 09:03:53 - |INFO| - [tb_gateway_remote_configurator.py] - tb_gateway_remote_configurator - __init__ - 73 - Remote Configurator started
tb-gateway  | 2024-11-06 09:03:53 - |INFO| - [tb_gateway_remote_configurator.py] - tb_gateway_remote_configurator - _process_config_request - 225 - Configuration update request received.
tb-gateway  | 2024-11-06 09:03:53 - |INFO| - [tb_gateway_remote_configurator.py] - tb_gateway_remote_configurator - _process_config_request - 225 - Configuration update request received.
tb-gateway  | 2024-11-06 09:03:54 - |INFO| - [statistics_service.py] - statistics_service - __send_statistics - 187 - REGULAR STATS: {'machineStats': {'totalCpuUsage': 12.9, 'freeMemory': 61.4, 'freeDisk': '409.4G', 'gwProcessCpuUsage': 20.0, 'gwMemory': 0.4, 'msgsSentToPlatform': 9, 'msgsReceivedFromPlatform': 4}, 'serviceStats': {'storageMsgPulled': 0, 'storageMsgCount': 0, 'platformMsgPushed': 0, 'platformAttrProduced': 0, 'platformTsProduced': 0}, 'connectorsStats': {}}
tb-gateway  | 2024-11-06 09:03:54 - |INFO| - [tb_gateway_remote_configurator.py] - tb_gateway_remote_configurator - _process_config_request - 225 - Configuration update request received.
tb-gateway  | 2024-11-06 09:03:54 - |INFO| - [tb_gateway_remote_configurator.py] - tb_gateway_remote_configurator - _process_config_request - 225 - Configuration update request received.
tb-gateway  | 2024-11-06 09:03:55 - |ERROR| - [tcp.py] - tcp - connect - 187 - Connection to (host.docker.internal, 5023) failed: timed out
tb-gateway  | 2024-11-06 09:03:58 - |ERROR| - [tcp.py] - tcp - connect - 187 - Connection to (host.docker.internal, 5024) failed: timed out
tb-gateway  | 2024-11-06 09:04:01 - |ERROR| - [tcp.py] - tcp - connect - 187 - Connection to (host.docker.internal, 5025) failed: timed out
tb-gateway  | 2024-11-06 09:04:04 - |ERROR| - [tcp.py] - tcp - connect - 187 - Connection to (host.docker.internal, 5026) failed: timed out
tb-gateway  | 2024-11-06 09:04:07 - |ERROR| - [tcp.py] - tcp - connect - 187 - Connection to (host.docker.internal, 5027) failed: timed out
tb-gateway  | 2024-11-06 09:04:10 - |ERROR| - [tcp.py] - tcp - connect - 187 - Connection to (host.docker.internal, 5028) failed: timed out
tb-gateway  | 2024-11-06 09:04:13 - |ERROR| - [tcp.py] - tcp - connect - 187 - Connection to (host.docker.internal, 5029) failed: timed out
tb-gateway  | 2024-11-06 09:04:16 - |ERROR| - [tcp.py] - tcp - connect - 187 - Connection to (host.docker.internal, 5030) failed: timed out
tb-gateway  | 2024-11-06 09:04:19 - |ERROR| - [tcp.py] - tcp - connect - 187 - Connection to (host.docker.internal, 5031) failed: timed out
tb-gateway  | 2024-11-06 09:04:22 - |ERROR| - [tcp.py] - tcp - connect - 187 - Connection to (host.docker.internal, 5032) failed: timed out
tb-gateway  | 2024-11-06 09:04:25 - |ERROR| - [tcp.py] - tcp - connect - 187 - Connection to (host.docker.internal, 5033) failed: timed out
tb-gateway  | 2024-11-06 09:04:28 - |ERROR| - [tcp.py] - tcp - connect - 187 - Connection to (host.docker.internal, 5034) failed: timed out
tb-gateway  | 2024-11-06 09:04:31 - |ERROR| - [tcp.py] - tcp - connect - 187 - Connection to (host.docker.internal, 5024) failed: timed out
tb-gateway  | 2024-11-06 09:04:34 - |ERROR| - [tcp.py] - tcp - connect - 187 - Connection to (host.docker.internal, 5023) failed: timed out
tb-gateway  | 2024-11-06 09:04:37 - |ERROR| - [tcp.py] - tcp - connect - 187 - Connection to (host.docker.internal, 5021) failed: timed out
tb-gateway  | 2024-11-06 09:04:40 - |ERROR| - [tcp.py] - tcp - connect - 187 - Connection to (host.docker.internal, 5022) failed: timed out
tb-gateway  | 2024-11-06 09:04:43 - |ERROR| - [tcp.py] - tcp - connect - 187 - Connection to (host.docker.internal, 5029) failed: timed out
tb-gateway  | 2024-11-06 09:04:46 - |ERROR| - [tcp.py] - tcp - connect - 187 - Connection to (host.docker.internal, 5030) failed: timed out
tb-gateway  | 2024-11-06 09:04:49 - |ERROR| - [tcp.py] - tcp - connect - 187 - Connection to (host.docker.internal, 5027) failed: timed out
tb-gateway  | 2024-11-06 09:04:52 - |ERROR| - [tcp.py] - tcp - connect - 187 - Connection to (host.docker.internal, 5031) failed: timed out
tb-gateway  | 2024-11-06 09:04:54 - |INFO| - [statistics_service.py] - statistics_service - __send_statistics - 187 - REGULAR STATS: {'machineStats': {'totalCpuUsage': 10.9, 'freeMemory': 60.8, 'freeDisk': '409.4G', 'gwProcessCpuUsage': 25.0, 'gwMemory': 0.4, 'msgsSentToPlatform': 21, 'msgsReceivedFromPlatform': 4}, 'serviceStats': {'storageMsgPulled': 0, 'storageMsgCount': 0, 'platformMsgPushed': 0, 'platformAttrProduced': 0, 'platformTsProduced': 0}, 'connectorsStats': {}}
```

**Versions**
 - OS: [Ubuntu 22.04]
 - Thingsboard IoT Gateway version [3.5.3]
 - Python version[3.10.12]

imbeacon, 2024-11-07:
Hi @KashifAhmd,

It looks like your connector tries to connect to host.docker.internal, but it cannot find running instance of server, do you have running instance of your Modbus server, that listen on requested ports?

imbeacon, 2024-11-15:
Closed due to inactivity.

closed 2024-11-15 by imbeacon